synthetos/g2 #124: G-Code Causes Machine Panic

ryansturmer, 2016-01-28:
```
G90
G20
G0X0.37500Y6.44250
G2X0.37500Y6.44250I0.G90F120
```

Causes our machine to panic, due to the bad number format of the I-word in the circle command.  I know this is bogus g-code, but a bad number format error is more appropriate than a machine panic in this instance.

I'm not sure it much matters what settings are used here - this pretty much universally brings our machines down.

aldenhart, 2016-02-18:
Yup. I see a NAN float from this.

aldenhart, 2016-02-23:
Resolution - the due to the malformed Gcode line "i" offset word is set to zero causing the offsets to be zero and the math to fail - leading to a panic. As of edge build 090.02 this condition is trapped and the status code "STAT_ARC_OFFSETS_MISSING_FOR_SELECTED_PLANE" (157) returned. 

The exception report and command response line are shown below:
{"er":{"fb":90.02,"st":157,"msg":"Arc specification error - missing offset(s) - arc offsets missing or zero"}}
{"r":{},"f":[1,157,29]}

Arc errors (such as this one) will also put g2 into an alarm state, rejecting any further commands until alarm is cleared with one of {clear:n}, {clr:n}, $clear, M30, M2

closed 2016-02-23 by aldenhart