synthetos/g2 #127: flashing tool can't access board in ubuntu

troisiemetype, 2016-03-04:
Hello,

I have  a problem flashing a due board in Ubuntu using terminal. Compiling goes without a problem, installing the bossa tool is fine too.
Activating bootlader is ok, I think, but I need to do it with sudo
But `bossac  --port=ttyACM0  -U true -e -w -v -i -b -R ./bin/gShield/gShield_flash.bin` gives `No device found on ttyACM0`
The strange thing is that n the arduino IDE, my due board appears with this port. But, arduino sketches can't be written to the board neither.
If I launch the arduno IDE with sudo, it works. But if I execute the bossac command with sudo, it makes no difference.

What should I do to have it work?
Thank you,

Pierre-Loup M.

troisiemetype, 2016-03-04:
Ok, sorry for asking questions I should find the answers by myself... After spending half a day on that, I found nothing.
But after having search more on the Internet about arduino problems, I finally found that the serial port can be accessed unless the user had been added to the tty and dialout groups.
`usermod -aG tty nomUtilisateur`
`usermod -aG dialout nomUtilisateur`

And by running the stty and bossac command with sudo, it's ok.

Sorry again for opening an issue for that!

Pierre-Loup.

aShure, 2016-03-18:
@troisiemetype Good point! I guess being in the dialout group is assumed to be a default if the Arduino IDE is installed and started at least once. Would you mind adding your fix to the flashing instructions? https://github.com/synthetos/g2/wiki/Flashing-G2-with-Linux

troisiemetype, 2016-03-30:
Hi,

I forgot to update the issue, but it has been added to the wiki. I think this issue can be closed.

closed 2018-04-18 by justinclift