synthetos/g2 #129: Support for CNC Shield

silopolis, 2016-03-18:
Hi,

Sorry if this is the wrong place to ask but... Is it possible to use a DUE running TinyG2 with a Protoneer's CNC Shield (http://blog.protoneer.co.nz/arduino-cnc-shield/) ? Or other boards ?

Thanks for your answer

aShure, 2016-03-18:
Yes that is possible.

silopolis, 2016-03-21:
hi @aShure , good news, thank you for your answer :-)

This is discussed ATM on Protoneer forum (see here http://forum.protoneer.co.nz/viewtopic.php?f=8&t=49).

Am I wrong this would first need to create a TinyG2 compile config for CNC shield with the appropriate pinout mappings ?! Is there an example that can be found somewhere for another board ?

But there seems to be power issues to be adressed before giving it a try...

Thanks again

closed 2017-01-04 by aldenhart

justinclift, 2018-05-07:
For anyone else coming across this, the "create a config for mapping pins and settings" wouldn't be hard.  The 5v vs 3.3v thing though is more worrying, as it sounds like this board could damage the Due over time in use.

On the other hand, if someone knows a way to modify the Protoneer CNC shield to use the 3.3v supply provided by the Due instead of the 5v one, it **might** work out ok.

Someone who knows electronics to a decent level (unlike myself atm) would need to look over it first.  No-one wants to destroy their stuff after all. :smile:

justinclift, 2018-05-07:
Hmmm, this looks relevant: https://forum.protoneer.co.nz/viewtopic.php?f=8&t=535 :smile:

justinclift, 2018-06-28:
Here's an approach to modify 5V Arduino shields to 3.3V, using a fairly simple voltage divider:

&nbsp; &nbsp; https://www.udoo.org/tutorial/arduino-shields-with-udoo/