synthetos/g2 #130: Homing behavior

troisiemetype, 2016-03-23:
Hello,

I've got a question/problem about homing behaviour. I've set up a cheap 40W laser cutter to work with TinyG2 on due. On this machine there are two switches, one per axis, that are used by the original board for homing.
I've set G2 to home on X-, Y+, according to the switch position. I've also set the Xmin, Xmax, Ymin, Ymax according to hardware dimension. So when homing, the X axis goes to its min, sets 0, then the Y axis goes to its max, and sets 0 too.
Is there a way for it to set the absolute value of the axis that is homed at its max to the max travel value?
On my router it's not a problem that Z+ is set at 0 as it's always set again on the top of the part to be route/milled, and it's logical: + is above the part, - is under.
But on the laser it makes it difficult to use it. So I home by calling
```G28.2X0Y0
G0Y-225
G28.3Y0```
but having it set directly to Ymax would be simpler. And, in this case, I think more obvious.

Thank you

Engineer3DPrototyping, 2016-05-11:
You can have the machine home to ymax on g2 but I don't think you can change it with out compiling you own version.
 Basically in the settings file you set the yaxis input pin to 4 and the homing direction to 1. If your using the gshield pinout. Be careful of edge_90 the only profile I could compile was the on for the shapeoko2 with the new junction aggression settings.

troisiemetype, 2016-05-11:
Thank you for this answer!
That's about what I've done: setting the y homing direction to 1, on input entry 4. But once homed the Y_MAX is set to 0, I would like it to be equal to Y_TRAVEL_MAX instead, with Y_TRAVEL_MIN = 0.
But that's no bgi deal since I use, without any problem, the small peace of code above to do homing...
I had also had problem when compiling anything else than Shapeoko2 settings: it appears that the two MANUAL_FEEDRATE_* parameters are missing in the other files.It took a while to find out this, but making binaries with verbose output gives the answer.

giseburt, 2016-05-30:
The next major push handles homing to max in this way: once joked it sets the axis position to the maximum travel value.

troisiemetype, 2016-05-30:
Thank you! I'll update it as soon as I can!

giseburt, 2016-12-05:
This was incorporated in 100.x push. Should have ben closed a while ago. The fix in in edge branch at the moment.

closed 2016-12-05 by giseburt