synthetos/g2 #132: Plans for an open serial (UART) port

adamsdv, 2016-04-09:
Hello, I'm embarking on a new machine and have, after careful study of the available options, decided that G2 is the control software I wish to invest in.  I will be running it on a DUE, probably using 2 gShields.  I'm considering other options at this moment though for the interface hardware.  I'm an electrical engineer who has spent the last 25 years building embedded systems (hardware and software) for various industries.

I want to plan now for foreseeable needs.  I initially missed the minor note that the new GPIO system, primarily the outputs, are not yet implemented.  I see the underpinnings and have found the other comments indicating that this code will soon be in the edge branch for use.  I anxiously await the next release with this code.

I have a few questions though.
I have successfully operated inputs 1-9 and shifted them to other pins for test purposes.
I see underlying code ready for 12 inputs,  I wonder whether you are planning to use this same GPIO mechanism to implement the feed-hold/cycle-start/reset type behaviors, many of which already have dedicated input pins allocated?

I saw a section on the wiki regarding multiple control interface ports, however with the DUE the native USB serial port (with two endpoints) will not suffice for my planned machine pendant.  I need a UART port.  As the Programming port uses UART 0 and the default limit switches for X/Y/Z take the other UART port pins, I will have to move at least one set of these (no problem) but what about getting the serial port and command parser connected to that UART?  Is this type of configuration something that you are planning for? 

I can only assume from the discussions of variable speed spindle control (And laser power control) using PWM/PFM/RS-485, that you are planning to add the ability to access the UARTS.  I'd recommend that configs be added to allow all the available UARTS to be accessed for differing functions.

How does this align with your plans?  What can I do to help?

giseburt, 2016-05-30:
The USART code is usable now in Motate in the edge-write-callbacks branch. In the next major push it will be used at an input source as well.

I'm not sure how it'll be used for spindle speed contro yet but that's on our roadmap so any suggestions are welcome.

guru54321, 2016-10-25:
I'm considering controlling my spindle using a RS485. I own a huanyang VFD and looking at the smoothie project, implementing such an interface appears to be not that hard. How to integrate something similar in g2core is another matter ofcourse. Unfortunately, Huanyang uses their own control codes, not compatible with modbus :-( .  See also the smoothie project page on spindle control : [http://smoothieware.org/spindle-module](http://smoothieware.org/spindle-module)
The nice thing about this is you can get actual feedback on spindle speed. So the gcode interpreter can wait for the spindle to speed up and once it reaches the set speed, continue instead of waiting a fixed dwell time (works too, but this is nicer)

closed 2023-02-16 by ril3y