synthetos/g2 #133: Serial interface no longer shows after flashing a Due on Mac OS 10.6.8

lincob, 2016-04-25:
Hi... This may not really be an issue, maybe just something in the "you're making it wrong" section...
I'm running OS X Snow Leopard (10.6.8)

I've flashed the bin found there : http://synthetos.github.io/g2/ (TinyG2_Due-edge-078.03-default.bin)

...On an Arduino Due, using bossac Version 1.3a... here is my output : 

```
MacBookPro:macosx nilocb$ ./bossac -p tty.usbmodem1d11 -e -w -v -b TinyG2_Due-edge-078.03-default.bin
Erase flash
Write 125860 bytes to flash
[==============================] 100% (492/492 pages)
Verify 125860 bytes of flash
[==============================] 100% (492/492 pages)
Verify successful
Set boot flash true
MacBookPro:macosx nilocb$
```

I press the reset button, i guess there is something happening because the RX pin starts to blink every 2 seconds or so (the RX pin close to the native port, not the one close to the programming port)...

When i plug back the Due native port on the mac, there is no more serial interface...

Do i need specific drivers ? Have i done something wrong ? Thanks for your help.

lincob, 2016-04-28:
Turned out to be a VCP FTDI drivers problem, which apparently require the 2.3 version (only available from OS 10.9 though)… So basically, on all previous OS versions, you’re stuck with 2.2.18 version drivers and it simply doesn’t work.

ril3y, 2023-02-16:
This is not an issue.

closed 2023-02-16 by ril3y