synthetos/g2 #134: Wrong setting in config

psykhon, 2016-05-25:
#define A_RADIUS (M1_TRAVEL_PER_REV/(2*3.14159628))
should be 
#define A_RADIUS (M4_TRAVEL_PER_REV/(2*3.14159628))
ans such

aldenhart, 2018-04-16:
Makes sense to make Motor4 the default.

Please keep in mind that some machines (like Othermill) use different motor/axis mappings for physical wiring reasons. I only bring that up here so it's brought to people's attention. We may want a comment that the setting is only a default. But perhaps that obvious by the filename :-)

justinclift, 2018-04-16:
Yeah, I'll create the PR for the settings_default.h file, but leave the others alone under the assumption they've already been tweaked for their intended purpose.

Which reminds me, I should look into the details of those other machines, then add them to the new "[Who uses g2core?](https://github.com/synthetos/g2/wiki/g2core-in-use)" page.