synthetos/g2 #135: Status outputs

hareedy, 2016-05-30:
Hey,
I'm trying to have couple leds indicating "cycle start" and feed hold" so i added these fuzzy function to canonical_machine.cpp 

void stat_out_leds(void)
{
	if (cm.hold_state == FEEDHOLD_HOLD){
	feedhold_pin.set();  <<< pin 55 on the Due 
	}
	else{
	feedhold_pin.clear();
	}
}

and called it back here in controller.cpp 
void controller_run()
{
	while (true) {
		_controller_HSM();
		stat_out_leds();                 <<<< 
	}
}

Works fine, but i'm not sure if it's the best place to call the function back.

giseburt, 2016-05-30:
It'll work for now. In the next major push we will have proper inputs and outputs in JSON. I hadn't thought about using thn outputs to indicate state before but it seems like a reasonable thing to support natively. I'll mark this as a feature request so it stays on the radar.

hareedy, 2016-05-30:
Thank you, also i'm working on adding physical controls (Cycle start, Pause, Resume) by setting the cm.hold_state and cm.cycle_state and i was worried about the calling location (PS. very very lousy coding skills here as i'm an electrical engineer :) ). any way thank you and waiting for the new push.

ryansturmer, 2016-08-31:
Yes, having an output that indicates "in cycle" is also valuable.

aldenhart, 2016-09-06:
Update: This is planned for the edge-100 branch but is not yet available. Please stay tuned.