synthetos/g2 #137: Hard limits

hareedy, 2016-06-02:
Hey again,
First i have to say it's the first time i use TinyG2, didn't even test it on a machine yet, just evaluating offline, any way, i don'r know if any of the guys came across this before, Ok, when tripping a limit switch the machine halts down, but the weird thing is when issuing a $clear command it goes to stop even if the limit still tripped and i dont think this is safe, also when resetting the board or restarting it and a limit is tripped the machine is in ready state and passing gcode normally. am i missing something?? is that normal??
Already modified the code to overcome this problems.

aldenhart, 2016-06-09:
Thanks for the update. Can you provide a pull request or a link so we can take a look at what you did? Thanks.

hareedy, 2016-06-09:
Glad to
[https://github.com/synthetos/g2/pull/140](url)