synthetos/g2 #139: Movement status not reported in text status reports

vonnieda, 2016-06-03:
Reporting this for a user of OpenPnP. Setting $sv=2 and issuing a movement command, the resulting status reports don't seem to include the "stat" keyword.

```
$sv=2
                                                                          
[sv]  status report verbosity     2 [0=off,1=filtered,2=verbose]
               
tinyg [mm] ok> 
                                                                
g0x0
                                                                           
tinyg [mm] ok> 
                                                                
line:0,posx:9.540,posy:0.000,posz:0.000,posa:0.000,feed:950.00,vel:364.42,unit:1
,coor:1,dist:0
                                                                 
line:0,posx:6.508,posy:0.000,posz:0.000,posa:0.000,feed:950.00,vel:984.37,unit:1
,coor:1,dist:0
                                                                 
line:0,posx:2.386,posy:0.000,posz:0.000,posa:0.000,feed:950.00,vel:886.01,unit:1
,coor:1,dist:0
                                                                 
line:0,posx:0.143,posy:0.000,posz:0.000,posa:0.000,feed:950.00,vel:172.03,unit:1
,coor:1,dist:0
```

I don't have access to G2 (until Sunday) so I can't verify this directly, but I asked the user to perform the command send me the results and this is what we got.

aldenhart, 2016-06-04:
It looks like the status report is not set up for the stat variable, which is what's needed. See here for how to set up status reports.

https://github.com/synthetos/TinyG/wiki/TinyG-Status-Reports#set-status-report-fields---firmware-build-44021-and-earlier

vonnieda, 2016-06-04:
Thanks @aldenhart. I've passed this on to the user. I have a Due arriving tomorrow and I'll be doing some testing of my own then.

closed 2016-09-06 by aldenhart