synthetos/g2 #143: TinyG2 to run a 5 axis CNC

Christine88, 2016-06-28:
Hi, so I built this small desk top 5 axis CNC and now I need to send gcode to it and control it.  My CNC has XYZ linear axis and A B rotary axis and 7 NEMA 17 stepper motors.  The A and B axis are driven by a dual stepper motor configuration using timing belts and 8:1 ratio while the XYZ axis are conventional lead screw configuration.  As far as I can tell based on Google searches, I need a TinyG2, Arduino Due and gShield but I'm really not sure.  If someone could advise me on this I would greatly appreciate it.

teshko, 2016-06-30:
AFAIK, gShield is only 3 axis, but you could use RAMPS 1.4 for smaller motors. RAMPS 1.4 is intended for use on Arduino MEGA2560 which is 5V and some pins need to be removed and connected to 3V3 pin. Look at schematic here: https://github.com/GermanRepRap/Ramps-1.4.2
You could also use: https://github.com/TopherMan/Ramps2Due

lutorm, 2016-09-11:
If you use a Due, there's really no problem using all 6 axes. Since you need external motor drivers anyway, just get as many as you need and hook them up to the stepper output pins on the Due.

closed 2023-02-16 by ril3y