synthetos/g2 #144: reading IO problem

siamect, 2016-07-17:
Dear All. 
Thanks for your fantastic efforts.
I have a problem reading inputs. If I do a read of all inputs {in:n} I can see all inputs (at least the first 9) but if I read a singel on {in2:n} I get zero (0) back:
```
{in:n}
{"r":{"in":{"1":0,"2":1,"3":0,"4":0,"5":0,"6":0,"7":0,"8":0,"9":0}},"f":[1,0,7]}

{in2:n}
{"r":{"in2":0},"f":[1,0,8]}

```
This probably leads to the problems I have with M101({in2:n})
I cannot find anything about M100 M101 in the current version of the Wiki but it is there in an older version.

thanks
Martin

siamect, 2016-07-19:
Checked the code for M100 M101 and it seems it's not implemented?...which makes a problem for me and I'm will to put some effort into getting it to work, but i need someone who knows the code to point me in the right direction.
Or are there any other ways of pausing the gcode execution with an input... synchronously with the gcodes so that I pause it in an exact point in the gcode stream? 
Any help is highly appreciated.

aldenhart, 2016-07-19:
You should be able to configure the IO system to perform a STOP action on a given pin. You would need to restart with a cycle start character (tilde, ~). Do not assign any function (none). 

See the GPIO page: https://github.com/synthetos/g2/wiki/Digital-IO-(GPIO) 

This is relatively new, and is in edge, so please let us know the results.

siamect, 2016-07-20:
Yes that seems to work as far as my short test goes. I sent 
`{di2ac:1}`
and the pis stops the current motion and a tilde continues it. 
I had at some case that it didn't continue an unfinished command when I bulled the pin, but I don't have anything to prevent glitches, it's just a loose cable and I touched it with my finger so I probably induced a burst.

This can be really useful, but what I really need is to run to a certain gcode and then wait until an input goes active and then continue the execution. Typically this can be used for temperature checking in 3D-printers or, as in my case. for a robot to wait for a piece to come in to a location so that I can pick it up.
Thanks for your efforts
/Martin

aldenhart, 2016-07-20:
Synchronizing the motion resume to external conditions will be supported once we release the M100 / M101 code. We are currently using these commands in a branch designed for 3d printing to synchronize motion to heater-at-temp, and we are also working to connect this to spindles that have at-speed signals.

siamect, 2016-07-21:
Fantastique! I'm interested in testing that branch :) is it published anywhere yet?

siamect, 2016-07-24:
Is there any possibility for me to get that branch?

aldenhart, 2016-09-04:
Back after a month of work... The branch I think you want is on edge-100. Try testing M101's with inN values. We are still testing all the functions, but try M101 ({in1:1}). Possibly M101 ({in1:t})  (or whatever input your are using if not 1). Not everything is wired up yet, but that should work.

closed 2016-09-06 by aldenhart