synthetos/g2 #146: how to add snap interrupt at 1khz rate in tiny-g2 for accurate machine position capturing need urgent help

shamsiqbal, 2016-07-30:
i am working on tiny-g2 but its very difficult code to understand so i thought that its better to ask that how to add 1khz interrupt in tinyg-2 for accurate machine axis position capturing 
here my idea is very simple capture motor axis position generated by dda ( at 1 khz interrupt rate genrated by sys-tick-timer or timer/counter channel) and send these absolute machine coordinate to other mcu for servo motor control loop doing this way i can save alot off hardware/software resources of second mcu where that controller have digital commanded position instead of step pulses counting and will be able to perform pid control loop for position control of each axis same function i have implemented in grbl running on stm32f103 mcu but i am looking for multi-axis(4-6) control option i am hopeful that someone will help me

aldenhart, 2016-08-21:
Note: This conversation moved to the TinyGdev group on Google.

closed 2016-08-21 by aldenhart

justinclift, 2018-07-30:
Ahhh, I think [this](https://groups.google.com/forum/#!topic/devtinyg/AYOu7hsrbn0) is the old discussion, which I hadn't found when looking previously.