synthetos/g2 #147: Single endpoint for USB communication ( this is not an issue )

sacidu93, 2016-08-07:
I developed a simple handheld terminal which is in theory capable of communicating with other hardware using one or more communication protocol available: USART, SPI, CAN, I2C, USB and Ethernet. I did some testing with GRBL on arduino uno using the USB bus in HOST CDC mode and it works just fine, the same test with TinygG with FT232 and it also works with some bugs. Now I want to do some testing with TinyG2 but here I'm not able to establish communication with the the ARM device in USB CDC mode, after some investigation I concluded that the problem is a software limitation of my USB HOST library that can't handle communication with an USB composite device.

I want to know if there is any way to modify the TinyG2 code in order to use a single CDC instance.

Any suggestion will be highly appreciated.
Thank you in advance.

[image]

[image]

[image]

giseburt, 2016-09-18:
Hi @sacidu93 ,

In the future single-CDC will be the default. (I can't guarantee that it will not be a composite device, as we may replace the secondary CDC with a mass-storage device.)

However, to make it a single-endpoint non-composite device at the moment (I'm assuming a Due + gShield configuration, but this will work for the others as well you just need to change the appropriate board's files):

Change [board_xio.h lines 36-39](https://github.com/synthetos/g2/blob/edge-100/g2core/board/ArduinoDue/board_xio.h#L36-L39) to only use one CDC:

```c++
extern Motate::USBDevice< Motate::USBCDC > usb;
// extern Motate::USBDevice<Motate::USBCDC, Motate::USBCDC> usb;
extern decltype(usb.mixin<0>::Serial)& SerialUSB;
// extern decltype(usb.mixin<1>::Serial)& SerialUSB1;
```

Then change [board_xio.cpp lines 43-47](https://github.com/synthetos/g2/blob/edge-100/g2core/board/ArduinoDue/board_xio.cpp#L43-L47) to match:
```c++
Motate::USBDevice< Motate::USBCDC > usb;
// Motate::USBDevice<Motate::USBCDC, Motate::USBCDC> usb;

decltype(usb.mixin<0>::Serial)& SerialUSB  = usb.mixin<0>::Serial;
// decltype(usb.mixin<1>::Serial)& SerialUSB1 = usb.mixin<1>::Serial;
```

The you need to make sure [xio.cpp](https://github.com/synthetos/g2/blob/edge-100/g2core/xio.cpp#L935-L962) doesn't try to use the now-missing `SerialUSB1`:

```c++
#if XIO_HAS_USB == 1
xioDeviceWrapper<decltype(&SerialUSB)> serialUSB0Wrapper {
    &SerialUSB,
    (DEV_CAN_READ | DEV_CAN_WRITE | DEV_CAN_BE_CTRL | DEV_CAN_BE_DATA)
};
// xioDeviceWrapper<decltype(&SerialUSB1)> serialUSB1Wrapper {
//     &SerialUSB1,
//    (DEV_CAN_READ | DEV_CAN_WRITE | DEV_CAN_BE_CTRL | DEV_CAN_BE_DATA)
// };
#endif // XIO_HAS_USB
#if XIO_HAS_UART==1
xioDeviceWrapper<decltype(&Serial)> serial0Wrapper {
    &Serial,
    (DEV_CAN_READ | DEV_CAN_WRITE | DEV_IS_ALWAYS_BOTH)
};
#endif // XIO_HAS_UART

// Define the xio singleton (and initialize it to hold our two deviceWrappers)
//xio_t xio = { &serialUSB0Wrapper, &serialUSB1Wrapper };
xio_t xio = {
#if XIO_HAS_USB == 1
    &serialUSB0Wrapper,
//    &serialUSB1Wrapper,
#endif // XIO_HAS_USB
#if XIO_HAS_UART == 1
    &serial0Wrapper
#endif
};
```

Make those changes, recompile and flash it and it should show only a single endpoint.

-Rob

aldenhart, 2017-01-04:
Closing this. As of the 100.15 builds there is only a single USB endpoint as a default. It is still possible to get 2 endpoints as a compile-time parameter by adding this line to the settings file: c

#define USB_SERIAL_PORTS_EXPOSED   2        // Valid options are 1 or 2, only!

closed 2017-01-04 by aldenhart