synthetos/g2 #148: Compile fails using settings_default.h, 'MANUAL_FEEDRATE_OVERRIDE_ENABLE' was not declared in this scope

yankee14, 2016-08-18:
I'm still having the exact issue described in #119.

$ make -j4 SETTINGS_FILE=settings_default.h
Make: /usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games:../Tools/gcc-arm-none-eabi/bin
Git: git
Looking for tools...
cd ../Tools && make "ARCH=gcc-arm-none-eabi"
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: Entering directory '/home/yankee/build/g2/Tools'
Compiling c CMSIS/Device/ATMEL/sam3xa/source/system_sam3xa.c
Compiling c CMSIS/Device/ATMEL/sam3xa/source/gcc/startup_sam3xa.c
Compiling c platform/atmel_sam/hooks.c
    -> build/gShield/CMSIS/Device/ATMEL/sam3xa/source/system_sam3xa.o 
    -> build/gShield/CMSIS/Device/ATMEL/sam3xa/source/gcc/startup_sam3xa.o 
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/yankee/build/g2/Tools'
    -> build/gShield/platform/atmel_sam/hooks.o 
Compiling c platform/atmel_sam/cortex_handlers.c
    -> build/gShield/platform/atmel_sam/cortex_handlers.o 
Compiling cpp motate/AvrUSB.cpp
    -> build/gShield/motate/AvrUSB.o 
Compiling cpp motate/SamUSB.cpp
    -> build/gShield/motate/SamUSB.o 
Compiling cpp motate/SamPins.cpp
    -> build/gShield/motate/SamPins.o 
Compiling cpp motate/SamSPI.cpp
    -> build/gShield/motate/SamSPI.o 
Compiling cpp motate/SamTimers.cpp
    -> build/gShield/motate/SamTimers.o 
Compiling cpp help.cpp
    -> build/gShield/help.o 
Compiling cpp cycle_jogging.cpp
    -> build/gShield/cycle_jogging.o 
Compiling cpp controller.cpp
    -> build/gShield/controller.o 
Compiling cpp planner.cpp
    -> build/gShield/planner.o 
Compiling cpp plan_arc.cpp
    -> build/gShield/plan_arc.o 
Compiling cpp hardware.cpp
    -> build/gShield/hardware.o 
Compiling cpp cycle_probing.cpp
    -> build/gShield/cycle_probing.o 
Compiling cpp plan_zoid.cpp
    -> build/gShield/plan_zoid.o 
Compiling cpp stepper.cpp
    -> build/gShield/stepper.o 
Compiling cpp plan_line.cpp
    -> build/gShield/plan_line.o 
Compiling cpp json_parser.cpp
    -> build/gShield/json_parser.o 
Compiling cpp cycle_homing.cpp
    -> build/gShield/cycle_homing.o 
Compiling cpp config_app.cpp
    -> build/gShield/config_app.o 
Compiling cpp kinematics.cpp
    -> build/gShield/kinematics.o 
Compiling cpp report.cpp
    -> build/gShield/report.o 
Compiling cpp xio.cpp
    -> build/gShield/xio.o 
config_app.cpp:486:98: error: 'MANUAL_FEEDRATE_OVERRIDE_ENABLE' was not declared in this scope
  { "sys","mfoe",_fipn,0, cm_print_mfoe,get_ui8, set_01,   (float *)&cm.gmx.mfo_enable,           MANUAL_FEEDRATE_OVERRIDE_ENABLE},
                                                                                                  ^
config_app.cpp:487:98: error: 'MANUAL_FEEDRATE_OVERRIDE_PARAMETER' was not declared in this scope
  { "sys","mfo", _fipn,3, cm_print_mfo, get_flt,cm_set_mfo,(float *)&cm.gmx.mfo_parameter,        MANUAL_FEEDRATE_OVERRIDE_PARAMETER},
                                                                                                  ^
Makefile:578: recipe for target 'build/gShield/config_app.o' failed
make: *** [build/gShield/config_app.o] Error 1
make: *** Waiting for unfinished jobs....`

Natekluver, 2016-08-19:
I too am having the same issue when compiling with Mac OSX.

Natekluver, 2016-08-20:
I did find if you comment out the below from TinyG2/config_app.cpp (line 486 and line 487) it compiles but I do not have it on a working machine yet to test. 
{ "sys","mfoe",_fipn,0, cm_print_mfoe,get_ui8, set_01,   (float *)&cm.gmx.mfo_enable,           MANUAL_FEEDRATE_OVERRIDE_ENABLE},
{ "sys","mfo", _fipn,3, cm_print_mfo, get_flt,cm_set_mfo,(float *)&cm.gmx.mfo_parameter,        MANUAL_FEEDRATE_OVERRIDE_PARAMETER},

aldenhart, 2016-08-21:
You can comment out those lines or define these parameters in the settings profile you are using
#define MANUAL_FEEDRATE_OVERRIDE_ENABLE     false
#define MANUAL_FEEDRATE_OVERRIDE_PARAMETER  1.00

closed 2016-08-21 by aldenhart