synthetos/g2 #149: some problems

taalabash, 2016-08-23:
I am using arduino due with a modified  pin assignment to suit my homemade cnc shield and i got these two problems :
1- I found that i have to connect the arduino to the pc two times or i have to reset the arduino to make the two com ports appear and to be able to send codes to the system , is that normal ??
 
2- when i send $di or $in the system sends me information about only 9 inputs not all the 12 inputs
I tried to modify the settings file to add settings to inputs 10 -12 with no chance
so I think I am missing something and I am using the master branch

so can any body help me please

lutorm, 2016-08-27:
(1) happens to me every time I've uploaded a new firmware, but otherwise it's detected right away. My suspicion is that it somehow gets stuck in the bootloader (http://playground.arduino.cc/Bootloader/DueBootloaderExplained) but reading through that page it's not obvious why that would be.

aldenhart, 2016-08-27:
System is set for only 9 as it thinks it's a v9 - that only has 9 inputs. We'll address this either in the edge push for build 100, or shortly after. It's a hard setting in the config_app.cpp table. The Hardware needs to be compiled for the Due to get past this.

taalabash, 2016-08-29:
Thank you lutorm for your answer

aldenhart :
after many times of trying to fix problem (2) and after searching in files , i think i fixed it finally by modifying this files
 config_app.h    gpio.cpp   gpio.h  tinyg2.h
and i have done that by adding lines for  inputs 10-12 similar to the 1-9 inputs lines and also by changing number of inputs and the number of homing enabled axis

thank you for help

closed 2016-09-01 by aldenhart