synthetos/g2 #153: Thunk on PowerUp

brookdrumm, 2016-08-30:
We would like the disconcerting "THUNK" to go away when you power the bot up.

giseburt, 2016-08-30:
This should be fixed since 33e13d1c10a9b67dedbadee0f0db420d117cda88. We should be able to test with release [100.02](https://github.com/synthetos/g2/releases/tag/100.02).

brookdrumm, 2016-08-30:
> This should be fixed since 33e13d1. We should be able to test with release 100.02.

Awesome!

giseburt, 2016-08-31:
Ok, so we have a deeper problem. Attached is a scope from the board, and the 3.3V rail isn't even up enough for the MCU to come online before the *thunk* happens.

[image]

I have verified that all *software* means of the startup jump have been removed, so I'm going to close this issue. We will need to track this further in the hardware's repo.

aldenhart, 2016-09-13:
This was software and hardware. The hardware fix was to fine tune the soft start time of the switching regulator so that the MCU has time to disable the drivers before the driver power-on sequence is completed.

closed 2016-09-13 by aldenhart

brookdrumm, 2016-09-14:
Really happy about this!! Great work.

We need the z axis timeout fixed on probing. Nick has details but rob probably knows what I am talking about.

Brook

ril3y, 2016-09-14:
Brooke the z-probing issue is better opened in a new issue or if its
already opened use a # then the number to reference it.  Either way it
helps us keep track of issues that are still open if they are on their own
issues.

Thanks!

giseburt, 2016-09-14:
@brookdrumm I've opened #166 to track the motor timeout bug.