synthetos/g2 #154: Temperature accuracy for hotend

brookdrumm, 2016-08-30:
The temperature we set is not accurate enough. 
Data: 
We set Nick's hotend to 180C and the thermocouple measured 200C, from inside the #2 thermistor hole on hotend.
Steve's was set to 180C and thermocouple reported 195C

giseburt, 2016-08-31:
A few questions to help narrow this down:
- What's the firmware reporting from each machine when it's at 180º?
  - Is the temperature mostly stable (+-1º) or is the firmware unable to maintain the temperature in roughly that range?
- If you swap the G2s, does the temperature reading move or stay the same for each extruder?
  - Said another way, if you currently have Nick's G2 reading Nick's extruder, and Steve's G2 reading Steve's extruder, what does Nick's G2 read from Steve's extruder, and the other way around? We're looking to see if the G2 is reading high or low or if the extruder reads high or low.
- Do we know what the tolerance is supposed to be on the thermistors in the extruders?

giseburt, 2016-08-31:
Upon re-reading the issue description, there are actually two issues here:
1. With a sample size of two, one of them is measuring 5ºC lower than the other while reporting the same temperature.
2. With a sample size of two, they are both measuring at least 10º higher than they they are reporting.

My previous comment was regarding problem (1), and is a variance from machine-to-machine (either the G2 or Extruder, TBD). For problem (2) we simply need to adjust the thermistor tables.

giseburt, 2016-09-07:
I adjusted the curves in 100.10-3-g59be2. @brookdrumm could you test and confirm an improvement (or not 😬 ).

aldenhart, 2016-09-15:
Completed. Accuracy is now within +/- 1 degree per thermocouple measurement

closed 2016-09-15 by aldenhart