synthetos/g2 #157: Integrate ABL Auto bed Level Feature

xpix, 2016-09-01:
Hi Friends,

i thought about a new feature, 3d Firmware i.e. reprap have an ABL Feature, they measure X-Points with a capacitive or inductive Sensor and . They use this to compensate the Bed surface to X-Axes. The received g-code will changed and this is a great feature for PCB milling or check surface bevor start milling.

here an example from the 3d print world:
http://www.instructables.com/id/Enable-Auto-Leveling-for-your-3D-Printer-Marlin-Fi/

Check also this Firmware code, he is great documented and maybe we can integrate some code segments:
https://github.com/repetier/Repetier-Firmware/blob/master/src/ArduinoAVR/Repetier/BedLeveling.cpp

hope u like this idea :)

giseburt, 2016-09-01:
Well, good news: "bed leveling" (a.k.a. "Tramming") is in the upcoming edge.

It works like this: you probe three times (with G38.2) then call {tram:1}
and it will set a 3D rotation matrix and offset to make the last three
probe points the Z plane.

We are running through and finding any last minute regressions and will
push this new code to edge ASAP.

-Rob

giseburt, 2017-01-10:
This has been integrated. I'm not sure why this hasn't been closed. :man_facepalming:

closed 2017-01-10 by giseburt