synthetos/g2 #159: the jtext command $outX=1 doesn't work

fca1, 2016-09-04:
(edge-100 branch)

When I try to change the value of an output,  ($out1=1, for example), there is no action for the output produce by due card.

(it seems like the function io_set_output (gpio.cpp) have a problem with the group (no empty) for parse the output_num,

giseburt, 2016-09-04:
The outputs for the gShield have not been assigned.

You can tell in
https://github.com/synthetos/g2/blob/edge-100/g2core/board/ArduinoDue/gShield-pinout.h#L224
where the outputs are assigned a -1.

Also note the higher in the file (
https://github.com/synthetos/g2/blob/edge-100/g2core/board/ArduinoDue/gShield-pinout.h#L73
) whether or the the output can PWM is decided. (This requires underlying
hardware support and that it doesn't require a timer or PWM channel that
otherwise used.)

Also check the documentation (
https://github.com/synthetos/g2/wiki/Adding-and-Revising-Motate-Pinouts )
for a complete work up on making a pin work.

-Rob

fca1, 2016-09-04:
(Edge-100)
Hello, 
Thank you very much for your answer but I am not agree with assignment of output, it's not the case, 

If I push a breakpoint inside the function io_set_output()  called by the parser, it appears than  the condition of the group is false and the argument X of $outX= is not correctly interpreted (strtol return 0), if this can help you,  

If the group "out" is replaced by "" (config_app.cpp) in the table of orders, I can set or clear the output with $outX=

Frantz

aldenhart, 2016-09-04:
This should be fixed in edge-100 build 100.09

closed 2016-09-04 by aldenhart