synthetos/g2 #160: Question about neopixel with sam3x8e target.

fca1, 2016-09-04:
Hello, 
We are very interested by the possibility to drive a neopixel ring, and we have tries to use with the target "gshield" and the PWM10 (pin10) to pilot the output ( sam3x8e ) 
Of course, hardware.cpp & hardware.h  are copied from printrboardg2 board.

The target is running and alls seems ok, but there is no signal on pin 10 (serial command of led's driver)

The DMA seems to start (function startTransfer() is called a first time), but pixel_pin.isTransferDone() is always false following that. 
Our question is : is the motate is compliance with that for the sam3x8e Target like the printrboardg2  ? 

Thank you in advance,

giseburt, 2016-09-04:
This isn't documented yet, it's very new and not very heavily tested.

In order for it to work without disrupting motion, we use the DMA
capability of the PWM peripheral. In order to not require two PWM channels,
you have to use a pin that's on PWM channel 0.

You can tell if it's defined as Motate::PWMTimer<0> in
https://github.com/synthetos/Motate/blob/sam4e-port/MotateProject/motate/Atmel_sam3x/motate_chip_pin_functions.h
.

Cross referencing that with
https://github.com/synthetos/g2/blob/edge-100/g2core/board/ArduinoDue/motate_pin_assignments.h
you can see this includes:
Pin 0, Pin 73, Pin 20, Pin 67, Pin 34, and Pin 35

We don't currently support using another pin. I hope this helps.

-Rob

giseburt, 2016-09-04:
I should add that the above pin list and places to look are specific to the
SAM3X8E.

fca1, 2016-09-05:
Thank you very much, the signal is provided on these pins, it's perfect.

aldenhart, 2016-09-05:
@fca1 - You are welcome. Would you mind contacting us at synthetos at gmail dot com? Not many people are using the Neopixels and we'd be interested in your results. Probably too long a discussion for Issues. We will report back what we learn so people can read it later. -- Thanks

closed 2016-09-18 by aldenhart