synthetos/g2 #163: Compile Error

Svendo1337, 2016-09-11:
Hi guys,

i have a problem when I want to compile edge-100...
I use AtmelStudio 7 and Win10x64.

Here are the Errors:
recipe for target 'win32/gcc-arm-none-eabi-5_4-2016q2-20160622-win32.zip' failed
recipe for target '../Motate/MotateProject/motate/../../Tools/win32/gcc-arm-none-eabi/bin' failed

Greetz

PS: I followed the tutorial on GitHub Wiki

simonemarin, 2016-09-17:
Same issue here.

giseburt, 2016-09-17:
In Atmel Studio there's an Output tab that shows more logging detail. Could you send the contents of that to help diagnose this?  Thank you!

Svendo1337, 2016-09-18:
Hi giseburt,
here is the log ^^
Thank you!

```
> ------ Skipped Build: Project: Motate, Configuration: Stub ARM ------
> Project not selected to build for this solution configuration 
> ------ Build started: Project: g2core, Configuration: gShield ARM ------
> Build started.
> Project "g2core.cppproj" (default targets):
> Target "PreBuildEvent" skipped, due to false condition; ('$(PreBuildEvent)'!='') was evaluated as (''!='').
> Target "CoreBuild" in file "D:\Program Files (x86)\Atmel\Studio\7.0\Vs\Compiler.targets" from project "C:\Users\tobia\g2\g2\g2core\g2core.cppproj" (target "Build" depends on it):
> 	Task "RunCompilerTask"
> 		Shell Utils Path D:\Program Files (x86)\Atmel\Studio\7.0\shellUtils
> 		D:\Program Files (x86)\Atmel\Studio\7.0\shellUtils\make.exe -C "C:\Users\tobia\g2\g2\g2core" -f "Makefile" BOARD=gShield CONFIG=ShapeokoDualY COLOR=0 VERBOSE=1 
> 		make: Entering directory 'C:/Users/tobia/g2/g2/g2core'
> 		"Found that we're in Atmel Studio"
> 		Das System kann den angegebenen Pfad nicht finden.
> 		Installing the necessary tools...
> 		cd ../Motate/MotateProject/motate/../../Tools && make "ARCH=gcc-arm-none-eabi"
> 		make[1]: Entering directory 'C:/Users/tobia/g2/g2/Motate/Tools'
> 		Das System kann den angegebenen Pfad nicht finden.
> 		The gcc-arm-none-eabi tools need installed. Downloading win32/gcc-arm-none-eabi-5_4-2016q2-20160622-win32.zip and installing now.
> 		mkdir -p win32 && cd win32 && curl -OL "https://launchpad.net/gcc-arm-embedded/5.0/5-2016-q2-update/+download/gcc-arm-none-eabi-5_4-2016q2-20160622-win32.zip"zip"
> 		Ein Unterverzeichnis oder eine Datei mit dem Namen "-p" existiert bereits.
> 		Folgender Fehler trat auf: -p.
> 		Ein Unterverzeichnis oder eine Datei mit dem Namen "win32" existiert bereits.
> 		Folgender Fehler trat auf: win32.
> C:\Users\tobia\g2\g2\g2core\Makefile(143,1): error: recipe for target 'win32/gcc-arm-none-eabi-5_4-2016q2-20160622-win32.zip' failed
> 		make[1]: *** [win32/gcc-arm-none-eabi-5_4-2016q2-20160622-win32.zip] Error 1
> 		make[1]: Leaving directory 'C:/Users/tobia/g2/g2/Motate/Tools'
> C:\Users\tobia\g2\g2\Motate\MotateProject\motate\Motate.mk(394,1): error: recipe for target '../Motate/MotateProject/motate/../../Tools/win32/gcc-arm-none-eabi/bin' failed
> 		make: *** [../Motate/MotateProject/motate/../../Tools/win32/gcc-arm-none-eabi/bin] Error 2
> 		make: Leaving directory 'C:/Users/tobia/g2/g2/g2core'
> 	Done executing task "RunCompilerTask" -- FAILED.
> Done building target "CoreBuild" in project "g2core.cppproj" -- FAILED.
> Done building project "g2core.cppproj" -- FAILED.
> 
> Build FAILED.
> ========== Build: 0 succeeded or up-to-date, 1 failed, 1 skipped ==========
> 
```

giseburt, 2016-09-18:
Inside the directory `Motate/Tools/` will be two new directories: `-p` and `win32`. Due to stupidity in the mkdir provided by AtmelStudio it fails of it attempts to create those and they already exist. (Actually, -p is the setting to tell it just that, and it's clearly not understanding and actually created that as a directory.)

If you delete those by hand it can get past that step.

However, it clearly failed after that step before so if it does please send along another log file. :-)

You will need to have a reliable internet connection (just this once) in order for it to download those tools. Once those are downloaded and stored in the `win32` directory you won't need to be online or download them again. If it fails during download you might get in this situation again.

I'll try to figure out why it's not using `gmkdir` on Windows like it should.

simonemarin, 2016-09-18:
Yes I ve also found out that it is missing Motate.. anyway I am on Ubuntu and I can see the same error.

giseburt, 2016-09-18:
Simone Marin,

If you're on Ubuntu, that's a whole different beast than Atmel Studio. If
you're missing Motate (and are on edge-100 branch) then see
https://github.com/synthetos/g2/wiki/Troubleshooting#problem-the-motate-directory-is-empty-the-motate-project-is-missing

lquiroz, 2016-10-13:
Svendo1337,

Due to stupidity of mkdir noted by giseburt I made some manual operations:
1. Download toolchain from https://launchpadlibrarian.net/268330601/gcc-arm-none-eabi-5_4-2016q2-20160622-win32.zip
2. Put zip file on .../Tools/win32
3. In a command prompt window cd to ...../Tools/win32 and execute ..\7za\7za x -ogcc-arm-none-eabi "gcc-arm-none-eabi-5_4-2016q2-20160622-win32.zip" (note "\"  and not "/ ". It's DOS !!)
4. Build solution again.  I finished with a g2core.elf

Regards

dumpguy, 2017-10-05:
Not sure if my problem is really the same, but I´d say it still exists.
Had errors:
"recipe for target 'win32/gcc-arm-none-eabi-5_4-2016q2-20160622-win32.zip' failed"
and
"recipe for target '../Motate/MotateProject/motate/../../Tools/win32/gcc-arm-none-eabi/bin' failed"

Found out the cloning of the g2 repository with GitHub Desktop (on Windows10) doesn´t download the win32 folder and files. Followed Iquiroz workaround, but had to rename the directory /arm-none-eabi to /gcc-arm-none-eabi.

dumpguy, 2017-10-05:
By the way, cloning the /Motate/ directory from git repo (located at https://github.com/synthetos/Motate) also doesn´t contain the /win32 folder.

dumpguy, 2017-10-07:
Found obviously the reason for my problems. After switching to the admin account on my PC, cloning once more the g2 repository and starting Atmel Studio 7 from scratch and origin 'g2core.atsln', the project build started as expected.

Relevant output as follows:

------ Build started: Project: g2core, Configuration: gShield ARM ------
Build started.
Project "g2core.cppproj" (default targets):
Target "PreBuildEvent" skipped, due to false condition; ('$(PreBuildEvent)'!='') was evaluated as (''!='').
Target "CoreBuild" in file "C:\Program Files (x86)\Atmel\Studio\7.0\Vs\Compiler.targets" from project "C:\Users\Admin\Documents\g2\g2core\g2core.cppproj" (target "Build" depends on it):
	Using "RunCompilerTask" task from assembly "C:\Program Files (x86)\Atmel\Studio\7.0\Extensions\Application\AvrGCC.dll".
	Task "RunCompilerTask"
		Shell Utils Path C:\Program Files (x86)\Atmel\Studio\7.0\shellUtils
		C:\Program Files (x86)\Atmel\Studio\7.0\shellUtils\make.exe -C "C:\Users\Admin\Documents\g2\g2core" -f "Makefile" BOARD=gShield CONFIG=ShapeokoDualY COLOR=0 VERBOSE=1 
		make: Entering directory 'C:/Users/Admin/Documents/g2/g2core'
		"Found that we're in Atmel Studio"
		The system cannot find the path specified.
		Installing the necessary tools...
		cd ../Motate/MotateProject/motate/../../Tools && make "ARCH=gcc-arm-none-eabi"
		make[1]: Entering directory 'C:/Users/Admin/Documents/g2/Motate/Tools'
		The gcc-arm-none-eabi tools need installed. Downloading win32/gcc-arm-none-eabi-5_4-2016q2-20160622-win32.zip and installing now.
		mkdir -p win32 && \
		cd win32 && \
		curl -OL "https://launchpad.net/gcc-arm-embedded/5.0/5-2016-q2-update/+download/gcc-arm-none-eabi-5_4-2016q2-20160622-win32.zip"
_[...]_
		mkdir -p win32 && \
		cd win32 && \
		../7za/7za x -ogcc-arm-none-eabi "gcc-arm-none-eabi-5_4-2016q2-20160622-win32.zip"
		7-Zip (A) 9.20  Copyright (c) 1999-2010 Igor Pavlov  2010-11-18
		Processing archive: gcc-arm-none-eabi-5_4-2016q2-20160622-win32.zip
_[...] extracting many files_
Everything is Ok
		Folders: 233
		Files: 5085
		Size:       432642321
		Compressed: 123123549
		make[1]: Leaving directory 'C:/Users/Admin/Documents/g2/Motate/Tools'
		gmkdir -p "build/ShapeokoDualY-gShield/motate/cmsis/TARGET_Atmel/sam3x/source/templates/"
_[...]_
Making binary bin/ShapeokoDualY-gShield/g2core.bin
		"../Motate/MotateProject/motate/../../Tools/win32/gcc-arm-none-eabi/bin"/arm-none-eabi-objcopy -O binary "bin/ShapeokoDualY-gShield/g2core.elf" "bin/ShapeokoDualY-gShield/g2core.bin"
		--- SIZE INFO ---
		"../Motate/MotateProject/motate/../../Tools/win32/gcc-arm-none-eabi/bin"/arm-none-eabi-size "bin/ShapeokoDualY-gShield/g2core.elf"
		   text	   data	    bss	    dec	    hex	filename
		 168424	      0	  35152	 203576	  31b38	bin/ShapeokoDualY-gShield/g2core.elf
		cp bin/ShapeokoDualY-gShield/g2core.elf g2core.elf
		cp bin/ShapeokoDualY-gShield/g2core.map g2core.map
		Build 100.26 100.26-4-g6519-dirty
		make: Leaving directory 'C:/Users/Admin/Documents/g2/g2core'
	Done executing task "RunCompilerTask".
	Using "RunOutputFileVerifyTask" task from assembly "C:\Program Files (x86)\Atmel\Studio\7.0\Extensions\Application\AvrGCC.dll".
	Task "RunOutputFileVerifyTask"
		
		Display Output File Size Skipped due to : Output File not found
	Done executing task "RunOutputFileVerifyTask".
Done building target "CoreBuild" in project "g2core.cppproj".
Target "PostBuildEvent" skipped, due to false condition; ('$(PostBuildEvent)' != '') was evaluated as ('' != '').
Target "Build" in file "C:\Program Files (x86)\Atmel\Studio\7.0\Vs\Avr.common.targets" from project "C:\Users\Admin\Documents\g2\g2core\g2core.cppproj" (entry point):
Done building target "Build" in project "g2core.cppproj".
Done building project "g2core.cppproj".

Build succeeded.
------ Skipped Build: Project: Motate, Configuration: Stub ARM ------
Project not selected to build for this solution configuration 
========== Build: 1 succeeded or up-to-date, 0 failed, 1 skipped ==========

justinclift, 2018-04-16:
This issue **seems** like it should be fixed with https://github.com/synthetos/Motate/commit/39e5aef705a809bfd8e91393b2f1d132ab15dd83.

The comment from @hodgeac though leaves me unsure.

@hodgeac Do you still do stuff with g2core, and did you get around to investigating the root cause of this in depth? :smile:

justinclift, 2018-04-18:
I've gone through the process of compiling `master` and `edge` branches on Win 7 x64 a few times using Atmel Studio 7 over the last day and a bit.

So far, no errors at all.  So I'll close this under the assumption the issue is now fixed.

If for some reason it turns out I'm wrong, please feel welcome to re-open it so we can investigate. :smile:

closed 2018-04-18 by justinclift