synthetos/g2 #164: G28 incorrect position feedback

lutorm, 2016-09-11:
I just tried to use G28.1 to define a position and then G28 to move to it. It appears there's something funny going on with the position feedback during a G28 move. Here's what happens:

* `g0 g53 x10 y-10 z-10`.
* `g28.1` now stores the position (10,-10,-10) in g53 coords.
* `g10 l20 p1 x20 y20 z20`. This same position is now (20,20,20) in g54 coords.
* `g54`
* `g0 x21 y21 z21`
* `g28`

The machine now moves to the correct position, (10,-10,-10) in g53. **However, it also reports that it is at position (10,-10,-10) in g54 coords!** This is displayed as the g54 coords in chilipeppr, and I think it's a G2 issue because the serial port JSON after the move says:

`{"sr":{"posx":10.000,"posy":-10.000,"posz":-10.000,"stat":3,"mpox":10.000,"mpoy":-10.000,"mpoz":-10.000}}`

I thought this might be how g53 moves are reported, but if I manually do a `g53 x10 y-10 z-10`, I get the correct feedback:

`{"sr":{"posx":20.000,"posy":20.000,"posz":20.000,"stat":3,"mpox":10.000,"mpoy":-10.000,"mpoz":-10.000}}`

justinclift, 2018-04-18:
Looks like this got missed somehow.  It definitely sounds like a bug.

Even though this was reported ~1.5 years ago, we should should definitely check if it's still happening in `master` and/or `edge` branches then figure what to do from there.

@lutorm Just to ask... since you're still doing stuff with g2core (eg judging by #323 :wink:), did you notice if this bug was still happening with edge in March? :smile:

aldenhart, 2018-04-18:
I will test this one, but I think it's been fixed. Let's leave it open.

lutorm, 2018-04-18:
No, I haven't looked lately. I don't have a lot of time for this right now but I can try it next time.