synthetos/g2 #165: How do G38.x probes work?

lutorm, 2016-09-12:
I'm trying to figure out if I could potentially use these, but I can't find anything about how they work. The LinuxCNC description says the machine position at probe switch is stored into parameters, but I don't know what that corresponds to in G2.

I found some indications from various people posting that seemed to say that executing a probe command *changes* the G53 machine system, but that seems really wrong to me. That's a "homing", not a "probe" operation. It would mean all the work coordinate systems would have to be recalculated, soft limits won't work, etc.

To be specific: what I'd really like to be able to do is to run a probe for tool length. This would specify what the position of the machine in the current work coordinate system should be at that point and change the tool length offset to make it so. If the machine just runs and stops at the position that trips the probe, that would work, since then I can issue a G10 L10 to the known probe position.

Can anyone point me to a description? If there's nothing around, I can dig into the code.

giseburt, 2016-09-12:
See:

https://github.com/synthetos/g2/blob/edge-100/g2core/cycle_probing.cpp#L106-L136 <https://github.com/synthetos/g2/blob/edge-100/g2core/cycle_probing.cpp#L106-L136>

And example of the stored probes being used is in the new {tram:1} command, defined here: https://github.com/synthetos/g2/blob/edge-100/g2core/canonical_machine.cpp#L449-L481 <https://github.com/synthetos/g2/blob/edge-100/g2core/canonical_machine.cpp#L449-L481>

I hope that helps!
-Rob

lutorm, 2016-09-12:
Thanks, that definitely helps. 

Does this "tramming" command mean that the infrastructure for supporting G68 coordinate system rotations also is in place now? It seems once you can define a rotation matrix it should be simple to add setting it with G68, too?

lutorm, 2016-09-12:
Also, how do you configure which input a probe will look at?

giseburt, 2016-09-12:
Yes, the same rotation matrix that’s being used for tramming could be used for G68.

The rotation matrix used is considerably more powerful, since it can do arbitrary 3D rotations.

See the rotation_matrix and rotation_z_offset being setup here: https://github.com/synthetos/g2/blob/edge-100/g2core/canonical_machine.cpp#L525-L538 <https://github.com/synthetos/g2/blob/edge-100/g2core/canonical_machine.cpp#L525-L538>

Also see canonical_machine_reset_rotation() https://github.com/synthetos/g2/blob/edge-100/g2core/canonical_machine.cpp#L711-L718 <https://github.com/synthetos/g2/blob/edge-100/g2core/canonical_machine.cpp#L711-L718>

Note that there’s only one matrix, so anything else that uses the matrix must be carful to add to the matrix (tricky!) or will not work with tram.

-Rob

giseburt, 2016-09-12:
That’s not done yet. Currently it’s hardcoded to the Z homing input: 

https://github.com/synthetos/g2/blob/edge-100/g2core/cycle_probing.cpp#L260-L261 <https://github.com/synthetos/g2/blob/edge-100/g2core/cycle_probing.cpp#L260-L261>

lutorm, 2016-09-12:
Aw... So you can't use probing and z-axis homing/soft limits at the same time, then?

giseburt, 2016-09-12:
No, it’s designed to do just that: Home Z, then Probe the bed with a Z-probe (same used in homing) three times, call {tram:1}, and now all motion will be tram to the (supposedly planar) bed.

-Rob

lutorm, 2016-09-12:
I guess I'm still confused... I thought typically homing was done with a fixed switch triggering off of something fixed on the machine, not based on the tool in the spindle, otherwise your machine coordinate system will depend on what tool you happen to have in the spindle when you home the machine. But the Z-probe has to touch the table so has to be done with the tool tip?

Is there an example of a machine using this setup somewhere?

giseburt, 2016-09-13:
Well, the Z-probe being used in this case is that of the PrintrBot Simple (a 3D Printer), which is a inductive sensor triggered by the still plate bed.

For CNC this would be either a probe switch placed in the toolhead or a touch plate on the bed with a clamp on the tool to complete the circuit.

I agree though that being able to change the probe input is helpful, and is something we’re working on.

-Rob

lutorm, 2016-09-14:
I played around with this and it seems I could indeed have a touch plate in-line with the homing switch, as long as the touch plate had a switch to short it for homing use. Thanks for clarifying this! But I would like to put in a vote for making the probe input configurable.

closed 2016-09-14 by lutorm