synthetos/g2 #167: Settings not being read

simonemarin, 2016-09-17:
Hallo
I am trying to compile 087 so I tweak settings_default.h with the values for my machine then I try to do just
make
or 
make PLATFORM=gShield SETTINGS_FILE=settings_default.h
but in both cases when I go to my machine (DUE) I can see that the values are not correctly set.
What am I missing here?
tx
Simone

aldenhart, 2016-09-17:
I'm not sure what's going on, but you can try overriding the SETTINGS_FILE setting in the make command line and using the default in the settings.h file. Which is - actually - settings_default.h. 

What OS are you compiling on?

simonemarin, 2016-09-18:
I think I ve found the problem, this is an extract of the Makefile:
```
# SETTINGS_FILE may get overriden by the PLATFORM setings below
SETTINGS_FILE ?= settings_default.h

#       SETTINGS_FILE="settings_othermill.h"^M
#       SETTINGS_FILE="settings_othermill_test.h"^M
#       SETTINGS_FILE="settings_probotixV90.h"^M
        SETTINGS_FILE="settings_shapeoko2.h"^M
#       SETTINGS_FILE="settings_shopbot_sbv300.h"^M
#       SETTINGS_FILE="settings_shopbot_test.h"^M
#       SETTINGS_FILE="settings_Ultimaker.h"

```

so it is compiling for the Shapeoko I remember the settings are those I see on the machine, can t try it right now but is seems pretty obvious. 
tx

closed 2016-09-18 by aldenhart