synthetos/g2 #171: Question: Chip Selection for a new board.

bdurbrow, 2016-10-12:
I would like to update my 8-axis board to support g2core, and would like to use the same chip that the v9 hardware will be using (I figure that doing so would be better supported than trying to do a port to a different processor myself).

Currently, the makefile for the v9 target seems to indicate that you are using a ATSAM3X8CA-AU.

So, my questions are: Should I use that chip? How committed to that chip are you for the v9 release? Or are you planning to move to a different chip?

I realize that v9 is still in development, so anything _could_ happen, but any guidance or outlook you could offer would be most appreciated.

:-)

aldenhart, 2016-10-29:
G2core supports many chips in the project now, including the ATSAM3X8CA-AU you mention. If you are going to 8 axes you might want to look at the ATSAM3X8E chip (also supported) 

G2core also supports other chips, including Atmel's M4 and M7 core series

shczm, 2017-03-09:
Does G2core support STM32F407 or STM32F427?

giseburt, 2017-03-09:
Not at the moment. If one were to add the support in Motate for it, then it would. :)

-Rob

closed 2023-02-16 by ril3y