synthetos/g2 #175: G2 sends extra step when starting arc

lutorm, 2016-10-22:
I was running a simple contour path and noticed that it sounded like a motor was losing steps right as it went to arc out of the contour. After some investigation, it indeed is losing steps and it does so even with an air cut, so it's not due to cutter force.

I hooked up the scope to the TinyG and it shows an extra step being introduced right as the arc starts. The G-code I ran was:

> G0 Z17.5
> G0 X75 Y2.313
> Z6.135
> G17
> G1 X35.42  F100.
> G3 X34.143 Y1.551 I0.015 J-1.474 F100.

By triggering the scope off the y-step output, I captured the following trace right as it started the final G3 move. The red and yellow are the X and Y step outputs, respectively:
[image]

The right vertical cursor is positioned on the first y-step. About 22ms earlier, at the left vertical cursor, there is an extra X-step introduced. This extra step must be why the motor is losing steps. Both the linear and arc moves have the same speed and are initially collinear, so there should be no change in the X-motor step stream as the arc starts.  This seems like a bug.

This is with my earlier fork of the edge branch, including tool length offsets. I haven't upgrade to edge-100 yet since (at least last I looked) the tool length offsets had not been added yet.

The full config dump is attached below.
[TinyG_config_161016.txt](https://github.com/synthetos/g2/files/545960/TinyG_config_161016.txt)

aldenhart, 2016-10-24:
Thank you for the detailed analysis. Can you please post the Gcode you used to reproduce the issue, and let us know the lines in question?

Also, you are on a pretty old build (89.03) I'd recommend checking out the current edge and trying this again, if you would be so kind. There have been a lot of changes since 89.03.

Edge is currently at 100.12. We are still bringing in new features such as some issues fixes and the tool offset code. the URL for the 100.12 tag is https://github.com/synthetos/g2/tree/100.12

Thanks

giseburt, 2016-10-24:
This looks like a different expression of #123. From your really detailed analysis (thank you!) the fix for that should fix this as well.

lutorm, 2016-10-24:
You're right, the symptoms sound very similar to #123, although there is no accel/decel in this case. I can't really run my machine without the tool offsets but I guess I can update the firmware temporarily to test whether it still happens.

giseburt, 2016-10-25:
I'm not sure how old your branch is, but you could maybe apply this patch
to see if it works:

https://github.com/synthetos/g2/commit/510f015a4b4faa1bd908f1ea32287f9dcab5707a

Better yet, update to edge (edge-100 is now in edge and master, and new
development is in edge).

-Rob

lutorm, 2016-10-26:
I manually applied that to my branch and it appears to have fixed the problem. I haven't scoped the outputs again but I can no longer hear the motor losing a step there.

closed 2016-10-29 by aldenhart