synthetos/g2 #176: LICENSE file at the top level

giseburt, 2016-10-25:


justinclift, 2018-04-11:
I can get this done pretty easily.  Which GPL license though, so I know which text to grab? :smile:

justinclift, 2018-04-12:
Ahhh, looking at a [source](https://github.com/synthetos/g2/blob/edge/g2core/gcode_parser.cpp#L750-L751) file now shows a GPL v2 (only) header.

aldenhart, 2018-04-16:
I ported this page over from the TinyG repo and modified it for g2core. 
https://github.com/synthetos/g2/wiki/G2core-License

It explains the intent of the licensing and the BeRTOS exception. 

I'll leave this issue open in case you think there's further action required.

justinclift, 2018-04-16:
Just created a PR adding it to the repo as a file, which I *think* will complete things for this issue. :smile:

justinclift, 2018-04-17:
License text has been merged into the top level of edge.

Is being merged into edge considered good enough for this issue to be closed, or should we generally wait until it's into master (more stable)?

justinclift, 2018-04-18:
After thinking about this a bit more... went and looked at the other recently closed PRs'.  Looks like being merged into edge is when things are considered gone (generally speaking).

So, closing this now. :smile:

closed 2018-04-18 by justinclift

giseburt, 2018-04-18:
Oh, good point. We somehow need to indicate that all development is done against edge, and all PRs then need to be against edge.

justinclift, 2018-04-18:
Well, the "do PR's and dev on edge" (paraphrasing) is mentioned a fair amount on the [Development process](https://github.com/synthetos/g2/wiki/Development-Process) page.

The bit I was unsure of was if getting into edge is considered "done" enough, or it the norm here is to wait for things to potentially stabilise.  eg wait for the new PR'd code to get into master

Seems to be "close the PR when merged into edge".  Which is completely fine.  Faster turn for PR's, and things could always be re-opened / new PR's created to follow up with anyway if needed.

Nothing to worry about. :smile:

giseburt, 2018-04-18:
No, that's a valid point. If we were using milestones, then I'd say *those* don't close until the PRs involved are merged to master. But we aren't quite there yet…

-Rob