synthetos/g2 #177: Clear queue after feedhold: unwanted motion

stg, 2016-10-26:
While running a job (planner queue filled), issue feedhold(!) then clear queue(%).
The axes stop at feedhold, but starts moving again after clear queue.

The unwanted motion is slow and appears to be towards some point later in the just cleared queue.

It does not, however, follow the previously queued path, rather moves along a linear path from the feedhold position to this unwanted end-position - crashing through anything that may be in the way.

I am running the dev-123-bad-decel branch, this may be related.
This behavious was never seen in pre edge-100 builds.

alessandrolecce, 2016-11-14:
My problem is that after ! If I send ~ the cycle doesn't restart, just write last position command.

aldenhart, 2016-11-14:
@stg , @alessandrolecce - How are you driving the feedhold and % character? Is this on the same channel or are you using 2 USB channels? What program are you using the send commands? Thanks

stg, 2016-11-14:
I am using Chilipeppr, same as always. Unsure, but I believe all is sent on the same channel still.

aldenhart, 2016-11-14:
OK. It is on the same channel.

aldenhart, 2016-11-16:
Brief update - we are working the issue and think we have located the culprit. A bit more testing to go.

alessandrolecce, 2016-11-16:
I work on a personal Interface write in C# on Windows.
I find the problem...It's because some command send answer so It's necessary  to read before send other command.