synthetos/g2 #178: Power management - gShield

antbarney, 2016-10-26:
I have tested with Version 100.11 and 100.12 but it seems no matter what settings are configured for the BOT the motors stay energized. Even using $md will not de-energize the motors. Am I missing something simple here? Attached is the settings file I used for both 100.11 and 100.12

[settings_cbeam.txt](https://github.com/synthetos/g2/files/553778/settings_cbeam.txt)

ril3y, 2016-10-27:
#define MOTOR_POWER_MODE            MOTOR_POWERED_IN_CYCLE  // default
motor power mode (see cmMotorPowerMode in stepper.h)

Means that when the axis is moving it will become powered.  Then after the
timeout (after it stops moving) it will de-energize.

Unless I understand what you mean by this?

Riley

antbarney, 2016-10-27:
That's correct, that's just so it assigns the same value to all 3 axis. Now the issue is regardless of movement the motors are being energized. They energize as soon as the Due is powered up. Another thing I found odd is even the cmd  $MD  (which if I read the info correctly) should turn off the motors did not do anything. It's like as soon as it's powered up it thinks it's in a cycle even with it doing nothing. And the machine state is "stopped".

-Anthony

From: Riley
Sent: Thursday, October 27, 8:50 AM
Subject: Re: [synthetos/g2] Power management - gShield (#178)
To: synthetos/g2
Cc: antbarney, Author

#define MOTOR_POWER_MODE MOTOR_POWERED_IN_CYCLE // default
motor power mode (see cmMotorPowerMode in stepper.h)

Means that when the axis is moving it will become powered. Then after the
timeout (after it stops moving) it will de-energize.

Unless I understand what you mean by this?

Riley

-

aldenhart, 2016-10-27:
Your motors are in ALWAYS ON mode pm=1. Please see this page:
https://github.com/synthetos/g2/wiki/Power-Management

It looks like the comments in the settings file are misleading. That could be the cause of the issue, and if so that needs to be corrected in the project.

antbarney, 2016-10-28:
I double checked my config and even did another build and verified the pm setting is set to 2
here is the output:
{"r":{"fv":0.99,"fb":100.12,"fbs":"100.12-2-g7b23","fbc":"settings_cbeam.h","hp":3,"hv":0,"id":"0084-d639-18c6-9ce","msg":"SYSTEM READY"},"f":[1,0,1]}
{"r":{"1pm":2},"f":[1,0,8]}
{"r":{"2pm":2},"f":[1,0,8]}
{"r":{"3pm":2},"f":[1,0,8]}

antbarney, 2016-11-04:
Any other idea's? I have walked thru the config files and can't seem to find anything that would point to there being a issue with the configuration.

aldenhart, 2016-11-04:
I ran the full regression and some manual testing and don't see the issues you are referring to. FYI, the file for the regression test is here - part of the tg_pytest suite:
https://github.com/synthetos/tg_pytest/blob/master/data/004-specials/powerManagement/power-management-fb100-001.json

I'm thinking that perhaps the pinout for the enable lines are misconfigured. What board profile and pin mapping are you using?

antbarney, 2016-11-04:
git submodule status
 1540f1af7fd2213a704fc8585513effae041f9f1 Motate (v0.9.0-134-g1540f1a)
from what I can tell I am using the latest Motate version.

I am using a Ubuntu setup to do the builds and am using am I missing a flag I should be adding for Pinout? this same command works perfecting from the master branch and the motors work as expected.

make BOARD=gShield SETTINGS_FILE=settings_cbeam.h (thats the config file I attached earlier)

giseburt, 2016-11-04:
Hi @antbarney,

I just figured out what's happening. In the refactor of the motor objects, I dropped support for the common enable pin that the gShield uses as an oversight.

I'll get that fixed ASAP. Sorry for the confusion.

-Rob

antbarney, 2016-11-04:
Awesome, glad it was an easy fix and I was not doing something completely noobish!

mennodegraaf, 2017-03-08:
Any updates on this one as I am experiencing this same issue with latest edge and a Protoneer CNC shield on top of the Due? With master it works fine.

The workaround suggested in #212 is not acceptable for me as I am using both GRBL compatible shields with common enable pin and custom designed with independent enable pin for each motor.

hurr1k4ne, 2017-03-08:
Just to Make Clear, the motor# enable Pins do Work as intended. 

But i am aware that those are Not usable for your Shield due to the Position of the Pins.

antbarney, 2021-09-01:
I no longer use g2 so closing this out

closed 2021-09-01 by antbarney