synthetos/g2 #179: Jogging with keyboard in chilipeppr hangs SPJS and appears to crash g2

lutorm, 2016-10-30:
I tried the new code (actually the dev-109-tool-offset branch) and when I try to use the keyboard to jog in chilipeppr, g2 appears to crash. The first move works normally, but when I press the key the second time, no motion happens. Afterwards, the motors are left energized.

The serial console in chilipeppr contained this, which ends with what looks like an exception. The first G0 is the one that works, the second one doesn't and get an "er" response.

> G91 G0 X1
> G90
> {"Lbs":4}
> {"r":{},"f":[1,0,10]}
> {"qr":47}
> {"Lbs":0}
> {"r":{},"f":[1,0,4]}
> !
> %
> 
> {"qr":""}
> {"Lbs":0}
> {"r":{"qr":47},"f":[1,0,10]}
> {"qr":48}
> {"sr":{"posx":1,"stat":3}}
> !
> %
> 
> {"qr":""}
> {"Lbs":0}
> {"r":{"qr":48},"f":[1,0,10]}
> {"sr":{"stat":3}}
> {"qr":""}
> {"Lbs":0}
> {"r":{"qr":48},"f":[1,0,10]}
> {"sr":{"stat":3}}
>  G91 G0 X1
>  G90
> {"Lbs":4}
> {"r":{},"f":[1,0,10]}
> {"qr":47}
> {"Lbs":0}
> {"r":{},"f":[1,0,4]}
> !
> %
> 
> {"qr":""}
> {"Lbs":0}
> {"r":{"qr":47},"f":[1,0,10]}
> {"er":{"fb":100.12,"st":201,"msg":"Move < min length - zoid() got L=0"}}
> {"er":{"fb":100.12,"st":201,"msg":"Move < min length - zoid() Vc=0 asymmetric HT case"}}
> !
> %
> 
> {"qr":""}
> {"qr":""}
>

aldenhart, 2016-11-01:
What is {"Lbs":0} and {"Lbs":4}? As far as I can tell that is not a valid JSON command. It returns null and an error: {"r":{"lbs":null},"f":[1,100,10]}

... as it should

Perhaps this is a question for John Lauer if these sequences originated with Chillippr

chilipeppr, 2016-11-01:
Lbs comes back from spjs to ChiliPeppr so the ui can be updated with char mode or line mode buffer data. So you can make sure things aren't getting out of whack.

antbarney, 2016-11-04:
I noticed this issue as well on 100.12 I enabled Verbose logging on SPJS and it looks like when you use the keyboard it sends this after the Jog command: send /dev/ttyACM0 !% {"qr":""} 

this is not present when you just press the button with the mouse. Could this be related? 

Verbose output from SPJS here.

using the keyboard:

2016/11/04 03:39:57 hub.go:122: sendjson {"P":"/dev/ttyACM0","Data":[{"D":"G91 G0 Z1\nG90\n","Id":"jog1"}]}
2016/11/04 03:39:57 serial.go:252: Done with writeJson method
2016/11/04 03:39:57 serialport.go:233: Got p.sendBuffered. data:G91 G0 Z1\n, id:jog1, pause:0
2016/11/04 03:39:57 bufferflow_tinygg2.go:192: BlockUntilReady(cmd:G91 G0 Z1, id:jog1) start
2016/11/04 03:39:57 bufferflow_tinygg2.go:225: New line length: 10, buffer size increased to:10
2016/11/04 03:39:57 serialport.go:233: Got p.sendBuffered. data:G90\n, id:jog1-part-2-2, pause:0
2016/11/04 03:39:57 bufferflow_tinygg2.go:192: BlockUntilReady(cmd:G90, id:jog1-part-2-2) start
2016/11/04 03:39:57 bufferflow_tinygg2.go:225: New line length: 4, buffer size increased to:14
2016/11/04 03:39:57 serialport.go:268: Got p.sendNoBuf. id:jog1, pause:0, data:G91 G0 Z1\n
2016/11/04 03:39:57 serialport.go:275: Items In SPJS Queue List:1
2016/11/04 03:39:57 serialport.go:268: Got p.sendNoBuf. id:jog1-part-2-2, pause:0, data:G90\n
2016/11/04 03:39:57 serialport.go:275: Items In SPJS Queue List:0
2016/11/04 03:39:57 bufferflow_tinygg2.go:309: < {"r":{},"f":[1,0,10]}
2016/11/04 03:39:57 bufferflow_tinygg2.go:345: Buffer decreased to itemCnt:1, lenOfBuf:4
2016/11/04 03:39:57 bufferflow_tinygg2.go:309: < {"qr":47}
2016/11/04 03:39:57 bufferflow_tinygg2.go:309: < {"r":{},"f":[1,0,4]}
2016/11/04 03:39:57 bufferflow_tinygg2.go:345: Buffer decreased to itemCnt:0, lenOfBuf:0
2016/11/04 03:39:57 hub.go:122: send /dev/ttyACM0 !% {"qr":""}
2016/11/04 03:39:57 serial.go:680: Inside spWrite arg: send /dev/ttyACM0 !% {"qr":""}
2016/11/04 03:39:57 serial.go:392: We need to manually pause our internal buffer.
2016/11/04 03:39:57 bufferflow_tinygg2.go:651: Paused buffer
2016/11/04 03:39:57 bufferflow_tinygg2.go:690: Found cmd that should skip buffer. cmd:!
2016/11/04 03:39:57 serial.go:414: Forcing this cmd to skip buffer. cmd:!
2016/11/04 03:39:57 serial.go:337: We got a command that is asking us to wipe the sendBuffered buf.cmd:%
2016/11/04 03:39:57 serial.go:368: Hit default in select clause
2016/11/04 03:39:57 serial.go:371: Done consuming sendBuffered cmds. ctr:0
2016/11/04 03:39:57 bufferflow_tinygg2.go:754: Lock being released in TinyG buffer
2016/11/04 03:39:57 bufferflow_tinygg2.go:867: Just sent release to b.sem so we will not block the sending to serial port anymore.
2016/11/04 03:39:57 serial.go:383: itemsInBuffer:1
2016/11/04 03:39:57 serial.go:401: We need to unpause our internal buffer.
2016/11/04 03:39:57 bufferflow_tinygg2.go:867: Just sent release to b.sem so we will not block the sending to serial port anymore.
2016/11/04 03:39:57 bufferflow_tinygg2.go:682: Unpaused buffer
2016/11/04 03:39:57 bufferflow_tinygg2.go:690: Found cmd that should skip buffer. cmd:%
2016/11/04 03:39:57 serial.go:414: Forcing this cmd to skip buffer. cmd:%
2016/11/04 03:39:57 serialport.go:268: Got p.sendNoBuf. id:, pause:0, data:!
2016/11/04 03:39:57 serialport.go:275: Items In SPJS Queue List:3
2016/11/04 03:39:57 serialport.go:268: Got p.sendNoBuf. id:, pause:0, data:%
2016/11/04 03:39:57 serialport.go:275: Items In SPJS Queue List:2
2016/11/04 03:39:57 serialport.go:233: Got p.sendBuffered. data:\n, id:, pause:0
2016/11/04 03:39:57 bufferflow_tinygg2.go:192: BlockUntilReady(cmd:, id:) start
2016/11/04 03:39:57 bufferflow_tinygg2.go:225: New line length: 1, buffer size increased to:0
2016/11/04 03:39:57 serialport.go:233: Got p.sendBuffered. data:{"qr":""}\n, id:, pause:0
2016/11/04 03:39:57 bufferflow_tinygg2.go:192: BlockUntilReady(cmd:{"qr":""}, id:) start
2016/11/04 03:39:57 bufferflow_tinygg2.go:225: New line length: 10, buffer size increased to:10
2016/11/04 03:39:57 serialport.go:268: Got p.sendNoBuf. id:, pause:0, data:\n
2016/11/04 03:39:57 serialport.go:275: Items In SPJS Queue List:1
2016/11/04 03:39:57 serialport.go:268: Got p.sendNoBuf. id:, pause:0, data:{"qr":""}\n
2016/11/04 03:39:57 serialport.go:275: Items In SPJS Queue List:0
2016/11/04 03:39:57 bufferflow_tinygg2.go:309: < {"r":{"qr":47},"f":[1,0,10]}
2016/11/04 03:39:57 bufferflow_tinygg2.go:345: Buffer decreased to itemCnt:0, lenOfBuf:0
2016/11/04 03:39:57 bufferflow_tinygg2.go:309: < {"sr":{"posz":0.97,"vel":124.78,"dist":1,"stat":6}}
2016/11/04 03:39:57 bufferflow_tinygg2.go:309: < {"er":{"fb":100.12,"st":201,"msg":"Move < min length - zoid() got L=0"}}
2016/11/04 03:39:57 bufferflow_tinygg2.go:309: < {"er":{"fb":100.12,"st":201,"msg":"Move < min length - zoid() Vc=0 asymmetric HT case"}}
2016/11/04 03:39:57 hub.go:122: send /dev/ttyACM0 !%
{"qr":""}
2016/11/04 03:39:57 serial.go:680: Inside spWrite arg: send /dev/ttyACM0 !%
{"qr":""}

2016/11/04 03:39:57 serial.go:392: We need to manually pause our internal buffer.
2016/11/04 03:39:57 bufferflow_tinygg2.go:651: Paused buffer
2016/11/04 03:39:57 bufferflow_tinygg2.go:690: Found cmd that should skip buffer. cmd:!
2016/11/04 03:39:57 serial.go:414: Forcing this cmd to skip buffer. cmd:!
[... 32 lines cut ...]
2016/11/04 03:39:58 serialport.go:275: Items In SPJS Queue List:0

Pressing the software button:

2016/11/04 03:46:08 hub.go:122: sendjson {"P":"/dev/ttyACM0","Data":[{"D":"G91 G0 Z1\nG90\n","Id":"jog2"}]}
[... cut ...]
2016/11/04 03:46:08 bufferflow_tinygg2.go:309: < {"qr":47}
2016/11/04 03:46:08 bufferflow_tinygg2.go:309: < {"r":{},"f":[1,0,4]}
2016/11/04 03:46:08 bufferflow_tinygg2.go:345: Buffer decreased to itemCnt:0, lenOfBuf:0
2016/11/04 03:46:08 bufferflow_tinygg2.go:309: < {"sr":{"posz":0.97,"vel":124.78,"dist":1,"momo":0,"stat":5}}
2016/11/04 03:46:08 bufferflow_tinygg2.go:309: < {"qr":48}
2016/11/04 03:46:08 bufferflow_tinygg2.go:309: < {"sr":{"posz":1,"vel":0,"dist":0,"stat":3}}
2016/11/04 03:46:13 bufferflow_tinygg2.go:309: < {"sr":{"stat":3}}

johnlauer, 2016-11-04:
The feed hold is crashing TinyG and thus SPJS appears (or does) lock up. I believe Alden now has this fixed after a session we did yesterday. Hoping we get a new release soon.

antbarney, 2016-11-04:
Ok, Good to hear. Being a total noob is there a reason that the Feedhold and queue flush only gets sent when using the keyboard?

chilipeppr, 2016-11-04:
Yeah, the on screen buttons are precise distance movements, while the
keyboard is variable depending on how long you press the key and when you
lift TinyG should stop immediately, thus the feed hold/wipe.

lutorm, 2016-11-04:
@chilipeppr @aldenhart Good to know you guys managed to track this down. I meant to try to figure out exactly what provoked it, but I've been busy with other things.

aldenhart, 2016-11-04:
Still testing, but yes, this is being worked

lutorm, 2016-12-06:
Any updates? Since it was unusable for me I can't upgrade from my year-old branch until this is fixed.

giseburt, 2016-12-06:
We are testing fixes to feedhold and queue flush in the dev-168-gquadratic
branch. Once that work is done then this should be working as well. If you
feel adventurous enough to give it a go we'd be happy to hear if it's
working for you.

The one part that needs more testing (and might still be broken) is the new
queue flush refactor. So if there's any problem it'll be there.

lutorm, 2018-03-23:
I tried the current edge and this seems to work now so this can probably be closed.

closed 2023-02-16 by ril3y