synthetos/g2 #180: High speed transmission bug

zerodamage, 2016-11-03:
When using a serial monitor to transmit G-code, the CTS/RTS flow control dos'nt seem to work, causing the machine to stall when lines are transmitted to fast. 

I see the g2 changelog mentions the issue but Motate changelog says that CTS/RTS should work?

zerodamage, 2016-11-10:
To iterate, it was possible previously to simply "spam" the controller with commands, and the RTS/CTS flow control would make sure that no planner overflow occurred; this is no longer possible.

giseburt, 2016-11-10:
CTS/RTS is functional and actively being used.

It’s posable to configure the board so that it doesn’t honor them, but none of our configurations that have a serial port do that.

What board are you on and what’s the configuration?

There may be another bug, unrelated to CTS/RTS, where if you don’t follow the protocol of sending a line when you receive an {r:…} it will cause the circular buffer to overflow. I’m still chasing that one down. That may be what you’re hitting.

-Rob

zerodamage, 2016-11-10:
After some further studying, I see that I have been confused between UART and USB-serial flow control. 

In (very old) version 078.03, it was possible to stream a g-file over USB-serial using CTS/RTS flow control only, however it seems to crash the controller on all newer versions.

aldenhart, 2016-11-17:
Update: This one is being worked. Please stay tuned.