synthetos/g2 #181: Compiling trouble

alessandrolecce, 2016-11-10:
I tried to compile the new firmware for arduino due in windows10 ...but I get this error and I'm not able to go on.

------ Inizio ricompilazione: Progetto: g2core, Configurazione: gShield ARM ------
Compilazione avviata.
Progetto "g2core.cppproj" (destinazioni: Clean):
Compilazione con la versione degli strumenti "14.0".
Destinazione "Clean" nel file "C:\Program Files (x86)\Atmel\Studio\7.0\Vs\Compiler.targets" dal progetto "E:\c#\Enere_Controller\g2core\g2-edge\g2core\g2core.cppproj" (punto di ingresso):
	Attività "RunCompilerTask"
		Parametro attività:Clean=True
		Parametro attività:
    ContentList=
        board\Archim\Archim-pinout.h
                SubType=compile
        board\Archim\board_stepper.cpp
                SubType=compile
        board\Archim\board_stepper.h
                SubType=compile
        board\Archim\board_xio.cpp
                SubType=compile
        board\Archim\board_xio.h
                SubType=compile
        board\Archim\hardware.cpp
                SubType=compile
        board\Archim\hardware.h
                SubType=compile
        board\Archim\motate_pin_assignments.h
                SubType=compile
        board\ArduinoDue\board_stepper.cpp
                SubType=compile
        board\ArduinoDue\board_stepper.h
                SubType=compile
        board\ArduinoDue\board_xio.cpp
                SubType=compile
        board\ArduinoDue\board_xio.h
                SubType=compile
        board\ArduinoDue\gShield-pinout.h
                SubType=compile
        board\ArduinoDue\hardware.cpp
                SubType=compile
        board\ArduinoDue\hardware.h
                SubType=compile
        board\ArduinoDue\motate_pin_assignments.h
                SubType=compile
        board\ArduinoDue\shopbotShield-pinout.h
                SubType=compile
        board\g2v9\board_stepper.cpp
                SubType=compile
        board\g2v9\board_stepper.h
                SubType=compile
        board\g2v9\board_xio.cpp
                SubType=compile
        board\g2v9\board_xio.h
                SubType=compile
        board\g2v9\g2v9k-pinout.h
                SubType=compile
        board\g2v9\hardware.cpp
                SubType=compile
        board\g2v9\hardware.h
                SubType=compile
        board\g2v9\motate_pin_assignments.h
                SubType=compile
        board\gquadratic\board_stepper.cpp
                SubType=compile
        board\gquadratic\board_stepper.h
                SubType=compile
        board\gquadratic\board_xio.cpp
                SubType=compile
        board\gquadratic\board_xio.h
                SubType=compile
        board\gquadratic\gquadratic-a-pinout.h
[... 761 lines cut ...]
		../Motate/MotateProject/motate/../../Tools/win32/gcc-arm-none-eabi/bin/arm-none-eabi-g++  -Wall -Wchar-subscripts -Wcomment -Wformat=2 -Wmain -Wparentheses -Wunreachable-code -Wsequence-point -Wreturn-type -Wswitch -Wtrigraphs -Wunused -Wuninitialized -Wno-unknown-pragmas -Wfloat-equal -Wundef -Wshadow -Wpointer-arith -Wwrite-strings -Wformat -Wmissing-format-attribute -Wno-deprecated-declarations -Wredundant-decls -Wlong-long -Wno-format-nonliteral -Wno-format-security -D__SAM3X8E__ -D__sam3x__ -D__ARM__ -mcpu=cortex-m3 -mthumb -ffunction-sections -fdata-sections -std=gnu++14 -fno-rtti -fno-exceptions -fstrict-volatile-bitfields -fdiagnostics-show-option  -DBOARD=gShield   -g3 -Os -I../Motate/MotateProject/motate/cmsis -I../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/include -I../Motate/MotateProject/motate/cmsis/TARGET_Atmel -I../Motate/MotateProject/motate/cmsis/TARGET_Atmel/preprocessor -I../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/include -I../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/source/templates -I../Motate/MotateProject/motate/Atmel_sam_common -I../Motate/MotateProject/motate/Atmel_sam3x -I../Motate/MotateProject/motate/platform/atmel_sam -I. -I../Motate/MotateProject/motate -I./board/ArduinoDue -Idevice/step_dir_driver -DMOTATE_BOARD="gShield" -DSETTINGS_FILE="settings_shapeoko2.h" -DMOTATE_CONFIG_HAS_USBSERIAL=1 -DGIT_VERSION=\"\" -DGIT_EXACT_VERSION= -MMD -MF build/ShapeokoDualY-gShield/dep/hardware.o.d -MT build/ShapeokoDualY-gShield/./board/ArduinoDue/hardware.o -xc++ -c -o build/ShapeokoDualY-gShield/./board/ArduinoDue/hardware.o ./board/ArduinoDue/hardware.cpp
		"Compiling cpp device/step_dir_driver/step_dir_driver.cpp"; echo "    -> build/ShapeokoDualY-gShield/device/step_dir_driver/step_dir_driver.o" 
		../Motate/MotateProject/motate/../../Tools/win32/gcc-arm-none-eabi/bin/arm-none-eabi-g++  -Wall -Wchar-subscripts -Wcomment -Wformat=2 -Wmain -Wparentheses -Wunreachable-code -Wsequence-point -Wreturn-type -Wswitch -Wtrigraphs -Wunused -Wuninitialized -Wno-unknown-pragmas -Wfloat-equal -Wundef -Wshadow -Wpointer-arith -Wwrite-strings -Wformat -Wmissing-format-attribute -Wno-deprecated-declarations -Wredundant-decls -Wlong-long -Wno-format-nonliteral -Wno-format-security -D__SAM3X8E__ -D__sam3x__ -D__ARM__ -mcpu=cortex-m3 -mthumb -ffunction-sections -fdata-sections -std=gnu++14 -fno-rtti -fno-exceptions -fstrict-volatile-bitfields -fdiagnostics-show-option  -DBOARD=gShield   -g3 -Os -I../Motate/MotateProject/motate/cmsis -I../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/include -I../Motate/MotateProject/motate/cmsis/TARGET_Atmel -I../Motate/MotateProject/motate/cmsis/TARGET_Atmel/preprocessor -I../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/include -I../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/source/templates -I../Motate/MotateProject/motate/Atmel_sam_common -I../Motate/MotateProject/motate/Atmel_sam3x -I../Motate/MotateProject/motate/platform/atmel_sam -I. -I../Motate/MotateProject/motate -I./board/ArduinoDue -Idevice/step_dir_driver -DMOTATE_BOARD="gShield" -DSETTINGS_FILE="settings_shapeoko2.h" -DMOTATE_CONFIG_HAS_USBSERIAL=1 -DGIT_VERSION=\"\" -DGIT_EXACT_VERSION= -MMD -MF build/ShapeokoDualY-gShield/dep/step_dir_driver.o.d -MT build/ShapeokoDualY-gShield/device/step_dir_driver/step_dir_driver.o -xc++ -c -o build/ShapeokoDualY-gShield/device/step_dir_driver/step_dir_driver.o device/step_dir_driver/step_dir_driver.cpp
		"Linking bin/ShapeokoDualY-gShield/g2core.elf" 
		"Using linker script: ../Motate/MotateProject/motate/cmsis/TARGET_Atmel/linker_scripts/sam3x/sam3x8/gcc/flash.ld" 
		../Motate/MotateProject/motate/../../Tools/win32/gcc-arm-none-eabi/bin/arm-none-eabi-g++  -T"../Motate/MotateProject/motate/cmsis/TARGET_Atmel/linker_scripts/sam3x/sam3x8/gcc/flash.ld" -L../Motate/MotateProject/motate/platform/atmel_sam -Wl,-Map,"bin/ShapeokoDualY-gShield/g2core.map" -o bin/ShapeokoDualY-gShield/g2core.elf -lgcc -lc  -g3 -Os -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -ffunction-sections -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -Wl,--entry=Reset_Handler -nostartfiles -mcpu=cortex-m3 --specs=nano.specs -u _printf_float -mthumb -L../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/source/gcc/     -Wl,--start-group build/ShapeokoDualY-gShield/./main.o build/ShapeokoDualY-gShield/motate/Atmel_sam_common/SamDMA.o build/ShapeokoDualY-gShield/motate/Atmel_sam_common/SamPins.o build/ShapeokoDualY-gShield/motate/Atmel_sam_common/SamPower.o build/ShapeokoDualY-gShield/motate/Atmel_sam_common/SamSPI.o build/ShapeokoDualY-gShield/motate/Atmel_sam_common/SamServiceCall.o build/ShapeokoDualY-gShield/motate/Atmel_sam_common/SamTimers.o build/ShapeokoDualY-gShield/motate/Atmel_sam_common/SamUART.o build/ShapeokoDualY-gShield/motate/Atmel_sam_common/SamUniqueID.o build/ShapeokoDualY-gShield/motate/Atmel_sam3x/SamUSB.o build/ShapeokoDualY-gShield/motate/platform/atmel_sam/syscalls.o build/ShapeokoDualY-gShield/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.o build/ShapeokoDualY-gShield/motate/cmsis/TARGET_Atmel/sam3x/source/templates/system_sam3x.o build/ShapeokoDualY-gShield/motate/cmsis/TARGET_Atmel/sam3x/source/templates/gcc/startup_sam3x.o build/ShapeokoDualY-gShield/motate/platform/atmel_sam/cortex_handlers.o build/ShapeokoDualY-gShield/motate/platform/atmel_sam/hooks.o build/ShapeokoDualY-gShield/motate/MotateUtilities.o build/ShapeokoDualY-gShield/motate/main.o build/ShapeokoDualY-gShield/./canonical_machine.o build/ShapeokoDualY-gShield/./config.o build/ShapeokoDualY-gShield/./config_app.o build/ShapeokoDualY-gShield/./controller.o build/ShapeokoDualY-gShield/./coolant.o build/ShapeokoDualY-gShield/./cycle_homing.o build/ShapeokoDualY-gShield/./cyc [... cut ...]
		make: *** [bin/ShapeokoDualY-gShield/g2core.elf] Error 1
		make: Leaving directory 'E:/c#/Enere_Controller/g2core/g2-edge/g2core'
	Esecuzione attività "RunCompilerTask" NON COMPLETATA.
Compilazione destinazione "CoreBuild" nel progetto "g2core.cppproj" NON COMPLETATA.
Compilazione progetto "g2core.cppproj" NON COMPLETATA.

Compilazione NON RIUSCITA.
========== Ricompilazione: 0 completate, 1 non riuscite, 0 ignorate ==========

After I get again 0.78 version...
Nobody can share the 1.00 version bin file???

Lenrockus, 2016-11-10:
https://github.com/synthetos/g2/releases

aldenhart, 2016-11-17:
Have you followed these instructions very carefully? The Windows10 / Atmel Studio 7 combination is very picky.

wedges, 2017-01-03:
I had the same problem.
The workaround I found was to protect the command in ..\g2\Motate\MotateProject\motate\motate.mk, line 448/450/452:
was: $(QUIET)$(NM) "$(OUTPUT_BIN).elf" > "$(OUTPUT_BIN).elf.txt"
after: "$(QUIET)$(NM)" "$(OUTPUT_BIN).elf" > "$(OUTPUT_BIN).elf.txt"
I did it for the next command and it's ok...
I'll try to find a real fix for it (for fun) and let you know.

(Sorry for a so late solution, I just came back here a looong time without updating my plateforme...)

giseburt, 2017-01-03:
@wedges That's no work-around, I think that's the solution!

Those commands should be in quotes anyway, so that was an oversight on my
part. I'll test that on the other platforms, by I think you found the
solution.

Thank you!
-Rob

wedges, 2017-01-03:
Great!
What I did not understand is why the previous ones were ok... That's why I called it workaround...

btw, soon i'll open another issue about the part downloading gcc-arm tools so you can commit the workaround/fix...
(mkdir -p is not good in windows...)

Regards,
Wedges

wedges, 2017-01-03:
The issue I talked about is linked to #163

dumpguy, 2017-10-06:
The present g2 edge repository still has a motate.mk with the above decribed issue, to be found in line no. 450: $(QUIET)$(NM) "$(OUTPUT_BIN).elf" >"$(OUTPUT_BIN).elf.txt"