synthetos/g2 #182: arduino due flash

alessandrolecce, 2016-11-10:
I find that the travel and step angle parameters aren't memorized in arduino due and always I need to resend the command from my C# interface on power-on.
Is it normal??

yck011522, 2016-11-10:
Yes. I think.
The developers haven't figured that part yet.

I usually compile the settings into the program so I don't need to send the settings every time.

alessandrolecce, 2016-11-10:
My problem is that I'm not able to compile....I open another post about it....always I arrive to to compile .elf but not the .bin!

aldenhart, 2016-11-17:
Have you followed these instructions very carefully? The Windows10 / Atmel Studio 7 combination is very picky.

giseburt, 2016-11-17:
The .elf file is copied to the top level on Windows since Atmel Studio requires it. The rest of the files (including the .bin) are in the bin/${target}/ directory, where ${target} will vary based on the make options you passed. (This allows you to build for several different types of board and with different settings without them clobbering the others.)

-Rob

alessandrolecce, 2016-11-30:
Unfortunately I get this error and I'm not able to remove it and compile successfully:

 
e:/c#/enere_controller/g2core/g2-100.12/motate/tools/win32/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe(0,0): warning: changing start of section .bss by 8 bytes
e:/c#/enere_controller/g2core/g2-100.12/motate/tools/win32/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe(0,0): warning: changing start of section .bss by 8 bytes
e:/c#/enere_controller/g2core/g2-100.12/motate/tools/win32/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe(0,0): warning: changing start of section .bss by 8 bytes
		"Exporting symbols bin/ShapeokoDualY-gShield/g2core.elf.txt" 
		../Motate/MotateProject/motate/../../Tools/win32/gcc-arm-none-eabi/bin/arm-none-eabi-nm "bin/ShapeokoDualY-gShield/g2core.elf" >"bin/ShapeokoDualY-gShield/g2core.elf.txt"
		".." i not internal command or an execute programm or a file batch.
		make: *** [bin/ShapeokoDualY-gShield/g2core.elf] Error 1

I'm becoming crazy! Please help me!

alessandrolecce, 2016-12-01:
Finally I found solution!
It's enough .elf file....then use instaction here:
http://atmel.force.com/support/articles/en_US/FAQ/Creating-bin-hex-file-from-elf-file-for-AVR-devices
Use vr-objcopy in Artmel install directorty non in Motate directory!

giseburt, 2016-12-01:
You don't need those instructions. (Those instructions are for AVR, and this is an ARM project.) The bin file is automatically made from the elf file at compile time.

Ignore the g2core.elf file dropped into the g2core directory by Atmel Studio. It's an artifact of the way it works.

Instead, use the files in `/bin/<configuration>/`, which will include an `.elf`, `.bin`, `.map`, etc. The `<configuration>` portion will change besed on the configuration you gave it.

See [Adding and Revising Boards → Output Files](https://github.com/synthetos/g2/wiki/Adding-and-Revising-Boards#output-files) for details.

alessandrolecce, 2016-12-01:
@giseburt 
I try two week to understand what happen when I compile the sulution in artmel studio, I get this error:
**************************************************************************
"Exporting symbols bin/ShapeokoDualY-gShield/g2core.elf.txt" 
		../Motate/MotateProject/motate/../../Tools/win32/gcc-arm-none-eabi/bin/arm-none-eabi-nm "bin/ShapeokoDualY-gShield/g2core.elf" >"bin/ShapeokoDualY-gShield/g2core.elf.txt"
		".." non è riconosciuto come comando interno o esterno,
		 un programma eseguibile o un file batch.
		make: *** [bin/ShapeokoDualY-gShield/g2core.elf] Error 1
E:\Csharp\Enere_Controller\g2core\g2-100.12\Motate\MotateProject\motate\Motate.mk(436,1): error: recipe for target 'bin/ShapeokoDualY-gShield/g2core.elf' failed
**************************************************************************
What's wrong in Motate.mk(436,1)????

alessandrolecce, 2016-12-02:
Anyway...I'm sorry...I used arm-objcopy.exe and not avr-objcopy.exe :)

giseburt, 2016-12-02:
I don't know why it's not working. It's literally just calling
arm-none-eabi-objcopy.

The only thing I can think of is to:

1) make sure Atmel Studio is at the latest possible version

2) remove the g2core/bin and Motate/Tools/win32 directory completely and
run the build again.

RoXX891, 2016-12-15:
Same Problem here... 

`"Exporting symbols bin/ShapeokoDualY-gShield/g2core.elf.txt" 
		../Motate/MotateProject/motate/../../Tools/win32/gcc-arm-none-eabi/bin/arm-none-eabi-nm "bin/ShapeokoDualY-gShield/g2core.elf" >"bin/ShapeokoDualY-gShield/g2core.elf.txt"
		Der Befehl ".." ist entweder falsch geschrieben oder
		konnte nicht gefunden werden.
C:\Users\Stefan\Documents\GitHub\g2\Motate\MotateProject\motate\Motate.mk(444,1): error: recipe for target 'bin/ShapeokoDualY-gShield/g2core.elf' failed`

It does not even get to the point to copy the binarys. The Problem is located in the "arm-none-eabi-nm" command in Line 448 of the Motate makefile.

It think there is a Problem with these toolchain files in the Motate/Tools/win32 Folder. If i remove them, it will not build at all. And also there is no download but i am connected all the time. I have to download these Files manually like it is mentioned here (the last post): https://github.com/synthetos/g2/issues/163

Anybody has some ideas?

alessandrolecce, 2016-12-15:
I gave up....
Create a .bat file with these code:

objcopy.exe -O binary g2core.elf g2core.bin

than transfer in the same folder g2core.elf generate from artmel studio and objcopy.exe from C:\Program Files (x86)\Atmel\Studio\7.0\toolchain\arm\arm-gnu-toolchain\arm-none-eabi\bin

So you can generate g2core.bin

Then you can transfer g2core.bin to arduino due with bossac.exe

justinclift, 2018-04-16:
@alessandrolecce @RoXX891 This issue is fairly old now, and had a possible fix added to the main source code near the start of 2017.

Are either of you still doing stuff with g2core, and if so did you get this working for yourselves? :smile:

justinclift, 2018-04-18:
Looking back at this issue, it seems like something that can be safely closed.

If for some reason it turns out I'm wrong, please feel welcome to re-open it so we can investigate. :smile:

closed 2018-04-18 by justinclift