synthetos/g2 #183: not getting information in chillipepper after hitting hard limit

AdoHaha, 2016-11-13:
I have a problem with hard limits. When I define one ( whether through .h file or json) and than 
push on the limit switch there is no info about this in Chillipepper (no alarm information), as it normally was ( with grbl or previous versions of g2). Therefore there is no reset button for alarm and I need to manually clear the alarm via the text command. 

Was there something changed? Is there a way to have this functionality back?

chilipeppr, 2016-11-17:
ChiliPeppr has never detected alarms from TinyG, but it sure would be nice. If you are up for it, the best thing to do would be to fork the TinyG widget which already watches all incoming JSON from TinyG. Then add to the code in the widget where it parses the incoming JSON and look for the alarm. If you see one perhaps throw up some kind of warning in the UI like above the TinyG widget. You'll likely have to create a reference list of the warning codes in the Javascript so you can give good explanations of what the alarm is from. Send me a pull request via Github and I'll publish it to the main widget.

AdoHaha, 2016-11-18:
Sure, seems doable.

Thanks for the reply and for the ChilliPeppr project!

closed 2016-11-18 by aldenhart