synthetos/g2 #184: I can't get spindle PWM to generate an output.

bdurbrow, 2016-11-13:
I'm trying to run g2 on an Arduino Due, with a custom shield to connect StepStick clone drivers. The axis stepper drivers work OK, but I can't get any PWM output on the spindle pin. It should be set up as Arduino pin 11.

I've tried disconnecting the shield and checking pin 11 with an oscilloscope  - the pin itself stays low.

I've attached the files that I've customized (a pinout header file, and a machine settings file, and added a CONFIG directive to boards.mk).

Any idea what I'm doing wrong?

[PWM Issue.zip](https://github.com/synthetos/g2/files/587829/PWM.Issue.zip)

aldenhart, 2016-11-17:
This most likely is a pinout mapping problem. Looks like you running an Arduino Due. Are you able to get spindle output using the gShield but not the step-stick shield?

bdurbrow, 2016-11-18:
That was one of the things I tried... no luck.

I've tried both:

make CONFIG=MaxNC10_160

... which is:

ifeq ("$(CONFIG)","MaxNC10_160")
    ifeq ("$(BOARD)","NONE")
        BOARD=gShield
    endif
    SETTINGS_FILE="settings_MaxNC10_160.h"
endif

... and just on the workbench (as obviously it's not a Zen machine, but it should be Ok to check for output with a DSO):

make CONFIG=DueZen7x12

... which is:

ifeq ("$(CONFIG)","DueZen7x12")
    ifeq ("$(BOARD)","NONE")
        BOARD=gShield
    endif
    SETTINGS_FILE="settings_zen7x12.h"
endif

giseburt, 2016-11-30:
I'm working on a testing branch for this. I'll update with the PR for you to test against shortly.

giseburt, 2016-11-30:
@bdurbrow Could you test branch dev-184-no-spindle-pwm and verify that it works? If so, I'll close this issue and merge in PR #191.

Thank you!

giseburt, 2016-12-15:
Merged with edge. THe code will be in tag 100.16

closed 2016-12-15 by giseburt