synthetos/g2 #185: Z probe errors before it even starts.

antbarney, 2016-11-15:
I have tried a number of things, and I fear I am just doing it wrong but as soon as I issue a G38.2 command to Probe to Zmin it immediately errors with the generic Probing Error. DI5MO is set to ACTIVE_LOW and is held high at 3.3v when tested. Any ideas why it would fail right away?

 G38.2 Z-100 F100
{"r":{},"f":[1,0,17]}
{"prb":{"e":0, "z":0.000}}
{"er":{"fb":100.12,"st":250,"msg":"Probe cycle failed - Probing error - probe failed to change."}}
{"sr":{"momo":4,"stat":2}}

 $$
 {"ej":""}
{"r":{"sys":{"fb":100.12,"fbs":"100.12-6-g41c8","fbc":"settings_cbeam.h","fv":0.99,"hp":3,"hv":0,"id":"0084-d639-18c6-9ce","jt":1.5,"ct":0.01,"sl":0,"lim":1,"saf":1,"m48e":1,"mfoe":0,"mfo":1,"mtoe":0,"mto":1,"mt":10,"spep":1,"spdp":0,"spph":1,"spdw":1,"ssoe":0,"sso":1,"cofp":1,"comp":1,"coph":1,"tv":1,"ej":1,"jv":2,"qv":0,"sv":1,"si":250,"gpl":0,"gun":1,"gco":1,"gpa":2,"gdi":0}},"f":[1,0,3]}
{"r":{"1":{"ma":0,"sa":1.8,"tr":8,"mi":8,"su":200,"po":1,"pm":2,"pl":1}},"f":[1,0,1]}
{"r":{"2":{"ma":1,"sa":1.8,"tr":8,"mi":8,"su":200,"po":0,"pm":2,"pl":1}},"f":[1,0,1]}
{"r":{"3":{"ma":2,"sa":1.8,"tr":8,"mi":2,"su":50,"po":0,"pm":2,"pl":1}},"f":[1,0,1]}
{"r":{"4":{"ma":3,"sa":1.8,"tr":360,"mi":8,"su":4.44444,"po":0,"pm":0,"pl":0}},"f":[1,0,1]}
{"r":{"x":{"am":1,"vm":8000,"fr":8000,"tn":0,"tm":420,"jm":500,"jh":1000,"hi":1,"hd":0,"sv":3000,"lv":100,"lb":4,"zb":2}},"f":[1,0,1]}
{"r":{"y":{"am":1,"vm":8000,"fr":8000,"tn":0,"tm":420,"jm":500,"jh":1000,"hi":3,"hd":0,"sv":3000,"lv":100,"lb":4,"zb":2}},"f":[1,0,1]}
{"r":{"z":{"am":1,"vm":4000,"fr":4000,"tn":-95,"tm":0,"jm":500,"jh":1000,"hi":6,"hd":1,"sv":2667,"lv":25,"lb":4,"zb":2}},"f":[1,0,1]}
{"r":{"a":{"am":0,"vm":360000,"fr":360000,"tn":-1,"tm":-1,"jm":22500,"jh":22500,"ra":1.273,"hi":0,"hd":0,"sv":180000,"lv":36000,"lb":5,"zb":2}},"f":[1,0,1]}
{"r":{"b":{"am":0,"vm":360000,"fr":360000,"tn":-1,"tm":-1,"jm":22500,"jh":22500,"ra":1.273,"hi":0,"hd":0,"sv":180000,"lv":36000,"lb":5,"zb":2}},"f":[1,0,1]}
{"r":{"c":{"am":0,"vm":360000,"fr":360000,"tn":-1,"tm":-1,"jm":22500,"jh":22500,"ra":1.273,"hi":0,"hd":0,"sv":180000,"lv":36000,"lb":5,"zb":2}},"f":[1,0,1]}
{"r":{"di1":{"mo":1,"ac":0,"fn":1}},"f":[1,0,1]}
{"r":{"di2":{"mo":1,"ac":0,"fn":1}},"f":[1,0,1]}
{"r":{"di3":{"mo":1,"ac":0,"fn":1}},"f":[1,0,1]}
{"r":{"di4":{"mo":1,"ac":0,"fn":1}},"f":[1,0,1]}
{"r":{"di5":{"mo":0,"ac":0,"fn":0}},"f":[1,0,1]}
{"r":{"di6":{"mo":1,"ac":0,"fn":1}},"f":[1,0,1]}
{"r":{"di7":{"mo":2,"ac":0,"fn":0}},"f":[1,0,1]}
{"r":{"di8":{"mo":2,"ac":0,"fn":0}},"f":[1,0,1]}
{"r":{"di9":{"mo":2,"ac":0,"fn":0}},"f":[1,0,1]}
{"r":{"do1":{"mo":1}},"f":[1,0,1]}
{"r":{"do2":{"mo":1}},"f":[1,0,1]}
{"r":{"do3":{"mo":1}},"f":[1,0,1]}
{"r":{"do4":{"mo":1}},"f":[1,0,1]}
{"r":{"do5":{"mo":1}},"f":[1,0,1]}
{"r":{"do6":{"mo":1}},"f":[1,0,1]}
{"r":{"do7":{"mo":1}},"f":[1,0,1]}
{"r":{"do8":{"mo":1}},"f":[1,0,1]}
{"r":{"do9":{"mo":1}},"f":[1,0,1]}
{"r":{"p1":{"frq":100,"csl":7900,"csh":12800,"cpl":0.13,"cph":0.17,"wsl":0,"wsh":0,"wpl":0.1,"wph":0.1,"pof":0.1}},"f":[1,0,1]}
{"r":{"g54":{"x":0,"y":0,"z":0,"a":0,"b":0,"c":0}},"f":[1,0,1]}
{"r":{"g55":{"x":0,"y":0,"z":0,"a":0,"b":0,"c":0}},"f":[1,0,1]}
{"r":{"g56":{"x":0,"y":0,"z":0,"a":0,"b":0,"c":0}},"f":[1,0,1]}
{"r":{"g57":{"x":0,"y":0,"z":0,"a":0,"b":0,"c":0}},"f":[1,0,1]}
{"r":{"g58":{"x":0,"y":0,"z":0,"a":0,"b":0,"c":0}},"f":[1,0,1]}
{"r":{"g59":{"x":0,"y":0,"z":0,"a":0,"b":0,"c":0}},"f":[1,0,1]}
{"r":{"g92":{"x":0,"y":0,"z":0,"a":0,"b":0,"c":0}},"f":[1,0,1]}
{"r":{"g28":{"x":0,"y":0,"z":0,"a":0,"b":0,"c":0}},"f":[1,0,1]}
{"r":{"g30":{"x":0,"y":0,"z":0,"a":0,"b":0,"c":0}},"f":[1,0,1]}

antbarney, 2016-11-15:
So, after some more testing the Z axis not moving was due to my limit switches. Now I can get Probing started but when using Zmin (DI5) it will not probe. But if I use Zmax (DI6) even tho this is configured as a Limit Switch the probe is successful. Is this by design? seems odd to me.

aldenhart, 2016-11-16:
Can you please post or make available settings_cbeam.h. It might be something in there.

antbarney, 2016-11-16:
Here is the settings file I am using, I have most everything working now as my limit switches where not connected correctly. The only issue I see now is it seems that the Probe operations are tied to the same Input as the Homing switch for the Axis. So I can't use the DI5 (Zmin) for the probe DI6 (Zmax) must be used. This seems to be different the older ways of configuring the axis you could specify both min and max switches but now it seems you can only assign one input for homing.

[settings_cbeam.txt](https://github.com/synthetos/g2/files/595152/settings_cbeam.txt)

aldenhart, 2016-11-17:
You are correct. There is only one homing input, as it's the one the machine moves towards. The probe is assigned to the Z homing switch. Line 256 of cycle_probing.cpp has a todo to make this assignable using the digital input function settings. I will close this issue and add an new issue to make the probe switch assignable.

closed 2016-11-17 by aldenhart