synthetos/g2 #186: Limit switch - Shutdown occurred

alessandrolecce, 2016-11-17:
What's the command to reset "Limit switch - Shutdown occurred" ??
I'm trying all command suggest in wiki without success.

aldenhart, 2016-11-17:
In either case a shutdown or an alarm may be cleared using {clear:n}. Either of these functions (or a panic) can be mapped to an input. In general, would wan to map limits to alarms, not shutdowns. Alarms will also clear at program end, which is M2 or M30.

Did you see this page:
https://github.com/synthetos/g2/wiki/Alarm-Processing

We have tried to make the wiki more navigable by including the right hand nav bar. This page is listed there as "Alarms and Exceptions" under Reference Pages. 

Please let me know if this solves the issue, or if there is a bug somewhere.

alessandrolecce, 2016-11-17:
I think these is a bug!
After alarm send alarm condition on serial : 
{"er":{"fb":78.02,"st":204,"msg":"Limit switch - Shutdown occurred","msg":cs1,"n":0,"gc":"$xsx=2"}}
but no command is accept!

aldenhart, 2016-11-17:
I think you are seeing normal operation. Please see here:
https://github.com/synthetos/g2/wiki/Alarm-Processing#alarm

What happens when an alarm or shutdown condition is hit is that all subsequent commands are rejected until the alarm or shutdown is cleared. This is for safety, so the machine does not crash or do anything unsafe. Some senders keep pounding the machine with Gcode even after an alarm is hit, so these must be rejected.

The 204 status code message you are receiving tells you that the system is in an alarm state. See here: https://github.com/synthetos/g2/wiki/Status-Codes

The command `{clear:n}` can be sent to clear the alarm, or an M2 or M30 will also clear it (end of job).

You can test this by manually setting and clearing an alarm:
g0 x1 (make sure the machine is functioning, should move to 1)
{alarm:true}
g0 x0 (will return a 204 status code)
{clear:null}
g0 x0 (will return to zero position)

alessandrolecce, 2016-11-17:
When I write "no command is accept" I want to tell that no command include $clear or M2 or M30 are accept!

aldenhart, 2016-11-18:
This message does appear in text mode. In JSON mode it does not because it assumes machine-to-machine communications. It may be a good idea to add the clear message as a "msg": to the JSOIN. Thanks for the suggestion.

from error.h:
static const char stat_204[] = "Command rejected by ALARM [$clear to reset]";
static const char stat_205[] = "Command rejected by SHUTDOWN [$clear to reset]";

alessandrolecce, 2016-11-30:
Unfortunately no command work.
Can some one help me? I use arduino due, with $xsx=2, $xsn=2 and so on.
When PIN 14-19 go to GND no command work to restart.
I'm trying to send command direct on serial without no interface.

alessandrolecce, 2016-11-30:
When shutdown occurs, RX led on arduino due start to blink continuously.

alessandrolecce, 2016-11-30:
Reading on https://github.com/synthetos/TinyG/wiki/Homing-and-Limits-Setup-and-Troubleshooting
I understood that the limit switch $xsx=2....(and so on) need reset, so all position are lost.
I need to use soft limit to use clear!
I find this absurd.
I think this can be sign like a bug.

alessandrolecce, 2016-11-30:
I think Digital Input limit is configurate as HALT and not like STOP!
How can i change it?

aldenhart, 2016-11-30:
Let's start at the beginning. You are on a very old firmware build. `{"er":{"fb":78.02,...}`

Please update to the current edge build. Also, what hardware are you running?

Lastly, we recommend using JSON and not the $ commands. These are equivalent, but the JSON commands send machine readable responses.

alessandrolecce, 2016-11-30:
Yes, my firmware is 78.02.
Problem is that no compiled firmware are avalable for "arduino due" (artmel ATSAM3X8E)....and I'm try to compile with Artmel Studio 7.0, but always I get problem with generation *.bin file, just *.elf are built.
I have a personal interface: I can read JSON and Text mode.

alessandrolecce, 2016-11-30:
Nobody can compile for arduino due and share .bin file?

hurr1k4ne, 2017-01-28:
@alessandrolecce the problem is, there is a lot of configuration that may not work with your setup.
So might be tricky...
I saw a lot of peole having problems with compilling in windows. 
They always advice to to read the instructions really careful. and i saw people writing that you may just rename the elf file to bin and upload it.
(please correct me if information is wrong)

closed 2023-02-16 by ril3y