synthetos/g2 #188: M101 question

jonbob0687, 2016-11-21:
Not sure if this is the right place to post/ask this. please let me know if i need to submit elsewhere.

I am excited about the new M100/M101 features. I have been able to use the following to blink an LED on an output pin on my DUE:
M100 ({out1:1})
G04 P0.5
M100 ({out1:0}) 
G04 P0.5
etc.
Awesome!!

I am trying to use this functionality to home a servo axis something like this:
M100 ({out1:1}) ;send io command to begin home
M101 ({in1:1}) ;wait for drive to say it has completed homing
G28.3 X0 ;set absolute position to zero to syncronize the drive with G2Core

I have tried the following:
M101 ({in1:1})
M101 ({in1:0})
M101 ({in1:n})

in any case, the program just plows through as though each line is just a comment and doesn't stop like i was expecting. I realize this is pretty fresh functionality, but i'm trying to understand if it's me or if the code just isn't fully operational yet.

setup:
Arduino Due
External servo pulse/direction drive
g2core edge branch compiled at version 0.99 build 100.12

chilipeppr, 2016-11-21:
I too was trying to get M101 to work to wait on an external ESP8266 Nodemcu to complete a task. Alden did a new firmware compile and put it up in the Github releases tab, but that now seems to have been taken away. Alden or Rob, do you know what happened to it? I was able to get M101 ({in1:0}) to work and it was amazing to see it happen. One thing that is missing though is a way to cancel the M101 command.

aldenhart, 2016-11-22:
I had to remove that release because of incompatibilities with the way we do tagging. I have posted binaries from that branch here. 
https://www.dropbox.com/sh/2589g64sx36nh8h/AABspiUXD2WR8T_22xeQD_Bta?dl=0

Please be aware that I have not been able to test these binaries as I am not at my shop right now (like really really not at my shop). So if there are issues I will need to address these after the Thanksgiving holidays.

siamect, 2016-12-08:
@aldenhart 
Any news about the M101 ({in1:0})  functionality yet?  
Or can you point me in the right direction where and how to solve this issue?
Thanks for a wonderful project!  
/Martin

aldenhart, 2016-12-08:
@siamect - We are wrapping up a major push of some new comms stuff to get the next generation board out, so that will be pushed soon. The branch I gave John for CP will be merged in shortly after. Possibly as early as this weekend if it passes all the regression tests.

I re-opened this issue so we can track your request.

siamect, 2016-12-10:
Ok great! Thanks!

jonbob0687, 2017-05-11:
Hi all,

After much reading through code in my case it was a simple front end error. 

In ChiliPeppr when the gcode was being loaded, it was pre-processing the gcode to add line numbers (N numbers). I guess this strips out any 'N#' from the file and re-numbers as it sees fit. so it stripped out the "n1" from the line M101 ({in1:t}) leaving M101 ({i:t}) which was subtle and not what i needed.

[image]

Make sure this is unchecked if you need to reference and input:
[image]

once i got past this i'm back to facing the 'real' issues. Kudos on adding this awesome feature!
Hopefully this is helpful to someone coming behind me.

PS. ChiliPeppr gets really scared when you try to run a gcode file without any motion (only IO calls).

chilipeppr, 2017-05-11:
Chilipeppr adds line numbers so you can see accurate reports on lines
executed. With no line numbers you get no information from tinyg. Also only
io calls is not chilipeppr acting weird. My gut is that chilipeppr just has
no idea what tinyg is doing cause no line numbers

<https://cloud.githubusercontent.com/assets/19519188/25926068/b075223e-35ba-11e7-8b89-1c66f7d01ea1.PNG>

<https://cloud.githubusercontent.com/assets/19519188/25926058/a1e23770-35ba-11e7-85dd-7d136ef4b116.PNG>

<https://github.com/synthetos/g2/issues/188#issuecomment-300645253>, or mute
the thread

closed 2023-02-16 by ril3y