synthetos/g2 #190: 6-axes from Due

siamect, 2016-11-30:
Dear all
I made a test long time ago to run 6-axes from an Arduino Due and after a little bit of fiddling around I got it to work. The version I started to work with at that time was a807727d7 "089.02 build for v9k testing".  
Now I tried to tweak a recent version (by coincidence 1bc31a6b) and the reporting back for axes a,b and c works but there are no pulses coming out... X,Y and Z works fine.  
Can anybody please point me in the right direction how to do this...  
Thanks for a fantastic project. 
/Martin

aldenhart, 2016-11-30:
Thanks for the heads up. That's a development branch. We will check it out. It's probably a pin mapping issue, so the place to look is in the Due / gShield pin mapping files.

alessandrolecce, 2016-12-01:
If you are not able to compile, the only available .bin file for Arduino Due is here: http://synthetos.github.io/g2/
it work without upgrade :)

siamect, 2016-12-01:
Thanks both of You. 
Compiling is no problem. I usually do it with `make BOARD=gShield ` which I assume use the settings_default.h file?
What worries me is that in the board_steppers.cpp and in the board_steppers.h the 5th and 6th motors are commented out. In the hardware.h the MOTORS are defined 4.  
May this be something I can just fix? (Tried and got compile errors):  
```
/home/martin/Documents/TinyG/g2core/./stepper.cpp:572: undefined reference to `motor_5'
/home/martin/Documents/TinyG/g2core/./stepper.cpp:572: undefined reference to `motor_6'
/home/martin/Documents/TinyG/g2core/./stepper.cpp:618: undefined reference to `motor_5'
/home/martin/Documents/TinyG/g2core/./stepper.cpp:618: undefined reference to `motor_6'

```
Thanks again for your support!  
/Martin

alessandrolecce, 2016-12-01:
Thanks.
I want to advice that I was able to compile in windows10 with Atmel studio 7.
After compile .elf file are generate, but not .bin file, so follows this instructions and you will be able to compile and flash into arduino due .bin file directly with bossac.exe (in arduino tool).

http://atmel.force.com/support/articles/en_US/FAQ/Creating-bin-hex-file-from-elf-file-for-AVR-devices
Use vr-objcopy in Artmel install directorty non in Motate directory!

siamect, 2016-12-01:
Alesandro, did you get pulses and direction out for all 6 axes on the Due?
What version did you compile and did you just uncomment the motor 5-6 in the  board_steppers.cpp and in the board_steppers.h and did you set the MOTORS to 6 in the hardware.h ?  
/Martin

alessandrolecce, 2016-12-01:
Yes.... be carfully in board.mk

ifeq ("$(CONFIG)","ShapeokoDualY")
    ifeq ("$(BOARD)","NONE")
        BOARD=g2v9k
    endif
    SETTINGS_FILE="settings_default.h"
endif

I found on original file not "settings_default.h" but another that I don't remenber!

giseburt, 2016-12-01:
@alessandrolecce This instruction are for AVR. The `.bin` file is made along with the hex file. See [Adding and Revising Boards → Output Files](https://github.com/synthetos/g2/wiki/Adding-and-Revising-Boards#output-files) for details.

Please ignore the `g2core.elf` file dropped in the `g2core/` directory, and only look at what inside `g2core/bin/*/`.

alessandrolecce, 2016-12-01:
I forget 
X_AXIS_MODE -> AXIS_STANDARD for all axis needed

@giseburt 
I understood where is the .elf file compiled.....the prooblem is that no bin file are generate when compile solution....so I used the artmel instructions....WORKS!

giseburt, 2016-12-01:
Hello @siamect,

There are a few places to change to get 6 motors going. For background, read through [Adding and Revising Boards → Making a New `BOARD`](https://github.com/synthetos/g2/wiki/Adding-and-Revising-Boards#making-a-new-board)

You're not adding one, just revising one.

So, generally, the step for adding a Motor are:
- Make sure the pins are correctly assigned. The Due/gShield configuration is backward in this respect, since we honor the Arduino pin numbering on the silkscreen. It appears that all of the motors are [pinned correctly](https://github.com/synthetos/g2/blob/edge/g2core/board/ArduinoDue/gShield-pinout.h#L135-L193), but you might verify.
- Make sure all of the motors are correctly configured in the [`board/*/board_stepper.h`](https://github.com/synthetos/g2/blob/edge/g2core/board/ArduinoDue/board_stepper.h) file.
  - In this case the two extra motors just need uncommented.
- Make sure that *same* configuration (yes, sadly there's some code duplication here) in [`board/*/board_stepper.cpp`](https://github.com/synthetos/g2/blob/edge/g2core/board/ArduinoDue/board_stepper.cpp).
  - `motor_5` and `motor_6` are commented out here too. Uncomment them.
  - Pay particular attention to the `Motors[MOTORS]` [initialization line](https://github.com/synthetos/g2/blob/edge/g2core/board/ArduinoDue/board_stepper.cpp#L86):
    ```c++
    Stepper* Motors[MOTORS] = {&motor_1, &motor_2, &motor_3, &motor_4};
    ```
  - You would need to add `motor_5` and `motor_6` to that list. Note that that's an array of pointers, so you need the `&`!
- `MOTORS` needs to be correct, and that's set in [`board/*/hardware.h`](https://github.com/synthetos/g2/blob/edge/g2core/board/ArduinoDue/hardware.h#L60)
  ```c++
  #define MOTORS 4       // number of motors on the board
  ```
  - Change that to `6` instead of `4`
- The configuration parameters are set in [`config_app.cpp`](https://github.com/synthetos/g2/blob/edge/g2core/config_app.cpp#L245-L268), and are keyed to `MOTORS`.
  - Note that if you go over 6, you'll have to adjust that file, and that's a whole other kettle of fish. See the comments in [`config_app.cpp`](https://github.com/synthetos/g2/blob/edge/g2core/config_app.cpp#L75-L103) for expalanataion.

siamect, 2016-12-01:
Thanks @giseburt  
I had only made the changes in the  board_stepper.h and not in the cpp file...  
Now it get through the build without errors.  
I'll test it in my robot tomorrow. It's midnight here so I need to sleep.  
Thanks a million everybody!  
/Martin

siamect, 2016-12-02:
Tested and it works. pulses comes out and the robot moves all axes.  
Thank again!   
/Martin

closed 2016-12-05 by giseburt