synthetos/g2 #195: Laser support (LaserWeb)

openhardwarecoza, 2016-12-06:
So the current level of support in lasers on other platforms is superb.  
Grbl 1.1e has beautiful support for G0=travel moves=beam off,  and G1=cut move =beam on
Smoothieware has the same method,  with the added bonus that pwm is adjusted on the fly as a function of acceleration (to prevent deeper engrave/darker etch in corners and direction changes.  At this Smoothieware won the open source laser firmware choice by a couple miles.  

So. In line with @giseburt 's request https://plus.google.com/+PetervanderWalt/posts/H9a6q27BtM6

Heres what we need
Laser mode on config
In this mode g0=pwm off,  g1=pwm on at last seen S value
S is usually inline with an G1 move for example G1 X10 Y10 F3600 S0.5 (a scale of 0-1 is used by smoothieware,  so 0.5=50% pwm) 
Pwm period / freq needs to be configurable too.  Diode lasers and co2 lasers use different frequencies for optimum linearity.  
If possible acceleration function in pwm level (proportional power down in slow down)

openhardwarecoza, 2016-12-06:
Note: already added basic comms to LaserWeb https://github.com/LaserWeb/LaserWeb3/commit/87876c6be51c5ae0da091585774b02684da263e2#diff-78c12f5adc1848d13b1c6f07055d996eR475

Just missing pwm control

cmaccioli, 2017-01-02:
Tried running G2core on Arduino Due (no shield) , but can only jog 1 increment 1 way and then back on each of axis, if I try to go more increments, G2 locks up.
I just got the system running and it works with chillipepper. 
Configured with a custom no-shield settings file, so Im wondering If I have something wrong there.

openhardwarecoza, 2017-01-02:
@cprezzi have to received your test board from @Giseburt yet?

cprezzi, 2017-01-03:
@openhardwarecoza Jep, got it 30 minutes ago :) and starting to tests now.

cprezzi, 2017-01-03:
@openhardwarecoza Do you was able to install the driver on Windows 10?

openhardwarecoza, 2017-01-03:
Yes it worked for me on win10.  But i might already have had a driver from
testing a V8 TinyG...  Maybe...

giseburt, 2017-01-04:
Ok guys, maybe you could clarify something for me. I see two primary types of lasers: "LED" (solid state) and "CO2". (I know there are many more, but these seem to be the popular ones in the "hobbyist" power range.)

The LED lasers appear to be purely pulse: on or off. Changing pulse duration is the only way of changing power output.

The CO2 lasers appear to have two controls. "Power" in the form of an analog signal which is controlling a constant current power to the tube itself, and "!Fire" which is what turns the laser on or off at the giver power level.

Is this a fair assessment so far?

openhardwarecoza, 2017-01-04:
Negative.

For your purpose,  dont worry about the wiring. Upon conversion,  some guys
keep the potentiometer,  some dont.  We have that scienced down to a T.
The CO2 PSU sums the pwm with the pot value internally,  no need to take
that into account in firmware.

We need a single PWM pin configurable with:
1) pwm period (kHz): different drivers respond best to different
frequencies.  200khz for some diode drivers,  76khz for some co2s, but
nothing set in stone. Some chinese diodes need 100khz. Its usually
documented in the drivers datasheet.  Make it a configurable value. .
Smoothieware handles it as: laser_module_pwm_period (see
http://smoothieware.org/laser)

2) invertable (some psus fire with a HIGH. Some with a LOW PWM Signal)

3) optional: an extra pin that can be used as a safety output (m3 enables
the laser psu. m5 disables the psu.  In between G0 stops pwm,  G1 Sx (sends
pwm,  G1S0 stops pwm)

giseburt, 2017-01-05:
Ok, so I did some more research on this.

For the purpose of setting general laser "power" we can use a PWM output
that is configurable. This would, in the CO2 case, be controlling the power
but not the pulses. On an LED those could be a reference output for the
constant-current power supply. This is likely not something that would be
set based on velocity or raster greyscale, since I suspect it would have
too high of latentency before it takes effect. In either case, we have pwm
output pins and will make them more configurable than they are now, and
it'll serve this purpose.

Now for the "fire" pin which is what actually controls the pulses, I see
there are two common schemes for controlling them.

  * "PPI" or pulses-per-inch. This is generally used for vector art and
cutting, and is actually two settings: "power" which is generally set as a
pulse duration, and the actual pulses-per-length itself. Note that it's
expected that given certain settings pulses may actually overlap (given the
circular area effected by a pulse), and this is intentional.
  * "Raster" mode. This mode is harder to get solid information on. A few
sources indicate that it is the same PPI scheme with a set pulse-per-length
setting and a varying pulse duration. I think that may be the easiest for
us to implement, and sounds like it should work reasonably well.

In this case, you would configure the machine with a given PPI, and then
configure the pulse duration indirectly with the S word. The pulse duration
would be set using a look-up curve converting the S word (from 0.0-1.0, for
example) to actual pulse duration times (it appears these can be measured
in milliseconds).

One interesting thing is that shorter pulses with more time in between will
have a higher spike of power, much like releasing the power from a
capacitive coil.

Thoughts?

-Rob

cmaccioli, 2017-01-10:
just from a very new user standpoint, is it possible to use a custom M code for laser and keep spindle (m3,m4,m5) standard. I am just starting, but have a cheap laser offset from spindle, so potentially could use both within same program if spindle and laser had separate PWM pins and M codes.  I.E. engrave in wood, then accent engraving with laser using a known offset. ? Just food for thought.
I know tinyg(G2) isn't ready for laser yet and am unsure how to do it, but when engraving lines with arcs, the machine slows in the corners, but laser power stays same, so corners/arcs are darker.  I apologize if this is way off track.

chilipeppr, 2017-01-10:
I'm interested in this too, but I did succeed using TinyG for laser by adjusting the jerk settings so that you don't get deceleration/acceleration, rather you get instant moves to get more of a continuous line. You just gotta play big time with the jerk.

giseburt, 2017-01-10:
@cmaccioli We plan on making the tool type configurable, and the tool
configuration will include choosing a toolhead. So, for example, you could
say T1 is the laser and T2 is a ¼" bullnose bit on the spindle. When T1 is
active the M3, etc control the laser. When T2 is active M3, etc control the
spindle. Does that sound like a reasonable solution?

cmaccioli, 2017-01-10:
First of all any kind of laser support for Tinyg (G2) would be great and I appreciate all the work everyone has done.
Regarding the Tool configuration,  would that be 2 separate Enable and PWM pins and configs, so you would have a pwm config for spindle (P1CPH, P1CPL, P1CSL, P1CSH.) and (P2CPH, P2CPL, P2CSL, P2CSH), or would you need to switch over your signals and configure for the laser driver or the spindle?

giseburt, 2017-01-10:
Each toolhead would be its own configuration. The laser configuration would
be different.

cmaccioli, 2017-01-10:
@giseburt - That sounds great, I can't wait. Thanks.

Mark-81, 2017-09-02:
Does the "laser mode" is now available in g2?

ril3y, 2023-02-16:
> Does the "laser mode" is now available in g2?

Yes but it needs testing.

closed 2023-02-16 by ril3y