synthetos/g2 #200: No Serial Port Visible

antbarney, 2016-12-15:
After doing a build for the gShield using the latest 100.14 when connecting up the Arduino Due I get the g2core heartbeat indicator but no Serial ports are exposed to Serial Port JSON server. I have tried from multiple machines and all show no Com ports.

Am I missing something here with the new USB changes in my build?

make BOARD=gShield SETTINGS_FILE=settings_cbeam.h is the command I am running.

giseburt, 2016-12-15:
Did you get the commits in edge from today? I fixed a similar issue with
USB on the Sam3x.

Make sure to update motate with `git submodule update`, since much of the
fix was over there.

Sorry about that. I should have opened an issue so you could see that this
was found and fixed. This just goes to show that when we don't issue an
issue for stuff we find ourselves, someone else will. :-)

-Rob

antbarney, 2016-12-15:
Yes I saw that there was a number of changes around the USB so made sure to pull in all of the commits from today my Motate is at commit edf4f30 which is the one listed as the current one for g2Core. I even blew away my local repo and re-cloned still the same issue. Attached is the bin file from my build.

[g2core_fail.zip](https://github.com/synthetos/g2/files/653538/g2core_fail.zip)

giseburt, 2016-12-15:
Ok, I found the issue. It's fixed in PR #201, so it'll be tracked there from here on.

Thank you for reporting this!

PR #201 (2016-12-15): Cleanup board files, and ensure board_hardware_init gets called.
Fixes #200
commit ff8524a6 (2016-12-15): Cleanup board files, and ensure board_hardware_init gets called.

Fixes #200
closed 2016-12-15 by giseburt