synthetos/g2 #202: g2command reject

alessandrolecce, 2016-12-17:
if start from x0y0 and send 
g2x0y0i1j0
command are reject.
Why?

giseburt, 2017-01-05:
Could you provide some more information? Hardware and configuration used, what exactly are you sending and what exact errors are you getting back?

alessandrolecce, 2017-01-12:
#223

alessandrolecce, 2017-01-13:
Arduino due

[fb]  firmware build             100.18
[fbs] firmware build ""
[fbc] firmware config "settings_default.h"
[fv]  firmware version            0.99
[hp]  hardware platform           3.00
[hv]  hardware version            0.00
[id]  g2core ID   0084-d639-18c6-18c
[jt]  junction integrgation time  0.75
[ct]  chordal tolerance           0.0100 mm
[sl]  soft limit enable           0 [0=disable,1=enable]
[lim] limit switch enable         1 [0=disable,1=enable]
[saf] safety interlock enable     1 [0=disable,1=enable]
[m48e] overrides enabled          1 [0=disable,1=enable]
[mfoe] manual feed override enab  0 [0=disable,1=enable]
[mfo]  manual feedrate override   1.000 [0.05 < mfo < 2.00]
[mtoe] manual traverse over enab  0 [0=disable,1=enable]
[mto]  manual traverse override   1.000 [0.05 < mto < 1.00]
[mt]  motor idle timeout          2.00 seconds
[spep] spindle enable polarity    1 [0=active_low,1=active_high]
[spdp] spindle direction polarity 0 [0=CW_low,1=CW_high]
[spph] spindle pause on hold      1 [0=no,1=pause_on_hold]
[spdw] spindle dwell time         1.0 seconds
[ssoe] spindle speed override ena 0 [0=disable,1=enable]
[sso] spindle speed override      1.000 [0.050 < sso < 2.000]
[cofp] coolant flood polarity     1 [0=low is ON,1=high is ON]
[comp] coolant mist polarity      1 [0=low is ON,1=high is ON]
[coph] coolant pause on hold      1 [0=no,1=pause_on_hold]
[tv]  text verbosity              1 [0=silent,1=verbose]
[ej]  enable json mode            0 [0=text,1=JSON,2=auto]
[jv]  json verbosity              2 [0=silent,1=footer,2=messages,3=configs,4=linenum,5=verbose]
[qv]  queue report verbosity      1 [0=off,1=single,2=triple]
[sv]  status report verbosity     1 [0=off,1=filtered,2=verbose]
[si]  status interval           250 ms
[gpl] default gcode plane         0 [0=G17,1=G18,2=G19]
[gun] default gcode units mode    1 [0=G20,1=G21]
[gco] default gcode coord system  1 [1-6 (G54-G59)]
[gpa] default gcode path control  2 [0=G61,1=G61.1,2=G64]
[gdi] default gcode distance mode 0 [0=G90,1=G91]