synthetos/g2 #204: TinyG G2 build 100.15 (edge) - board not responding after '$$'.

Tegeon, 2016-12-22:
I have loaded latest G2 edge firmware build 100.15 and found that board cannot be used in Chilipeppr TinyG workspace. Connecting a terminal has shown that board responds to various status commands ($, $h. $1, $2, $3, $4, $x, $y, $z, $a, $b, $c etc.) but as quick as $$ is being sent after few output lines no communication is possible anymore until RESET. There is a G+ post here: https://plus.google.com/+SebastianSzafran/posts/Kfqr86mTevo describing the problem in details.

hurr1k4ne, 2016-12-22:
I can confirm the issue. 
Generic Arduino Due + synthetos gShield V5

working with release 87.* working.

g2core 100.15 compiled on ubuntu 16.10, make parameters:" make BOARD=gShield CONFIG=default"
flashed with bossac over program port

Neither with one nor with two ports exposed. even with or without UART silenced.

aldenhart, 2016-12-23:
Being worked

cmaccioli, 2016-12-23:
Newbie having same issue here.
Trying to run due with no shield, external stepper drivers.
Also had to mess with the tinyg.inf and manually install it on win 8 to get driver to even see a port, 
On a side not I tried to compile with a different settings file, but the $ reports settings_default.h, is that normal?

aldenhart, 2016-12-24:
This has been fixed and pushed to edge. Build number s still 100.15 until we get some testing.

cmaccioli, 2016-12-27:
Now working with chillipeppr logo in chillipepper.
haven't done much other testing.  running on due with no shield to external steppers @ 3.3v.
Only issue is spindle speed and stepper enable signals need to be inverted (is there a way to invert stepper enable signals in software?).

aldenhart, 2017-01-03:
Fixed in build 100.16

closed 2017-01-03 by aldenhart