synthetos/g2 #205: Build for Due/gShiled

perfectphase, 2016-12-23:
Hi, in advance apologies if this is the wrong place to ask this.

I'm trying to build (for the first time) g2 for Due+gShield.  I've tried this on Windows10+Bash and Ubuntu 16.4.1 and have failed on both.

Basic steps:
sudo apt-get install git-core make
git clone https://github.com/synthetos/g2.git --recursive
cd g2/g2core
make CONFIG=TestV9

On the Ubuntu machine failed at:

make[1]: Leaving directory '/home/stephen/g2/Motate/Tools'
Compiling c ../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.c
    -> build/TestV9-g2v9k/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.o
bash: ../Motate/MotateProject/motate/../../Tools/linux/gcc-arm-none-eabi/bin/arm-none-eabi-gcc: No such file or directory
../Motate/MotateProject/motate/Motate.mk:473: recipe for target 'build/TestV9-g2v9k/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.o' failed
make: *** [build/TestV9-g2v9k/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.o] Error 127

The Windows10+Bash build fails at:
"Found that we're in Linux"
Compiling c ../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.c
    -> build/TestV9-g2v9k/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.o
bash: ../Motate/MotateProject/motate/../../Tools/linux/gcc-arm-none-eabi/bin/arm-none-eabi-gcc: cannot execute binary file: Exec format error
make: *** [build/TestV9-g2v9k/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.o] Error 126

Could any point me in the correct direction?

Also is it correct that to build for the Due+gShield I need to do: make BOARD=gShield

Thanks

Stephen.

giseburt, 2017-01-02:
The arm toolchain we are currently using is built for 32-bit on linux, so you'll need the 32-bit compatibility libraries installed on your version of linux for them to work. They are installed differently for the various version of linux, but should be easy to install.

Note: the next version of the tools (which we will switch to once tested) are build for 64-bit on linux. This means they no longer support 32-bit machines on linux. There's not a lot we can do about that, unfortunately.

@perfectphase please let us know if that solves your problem.

-Rob

perfectphase, 2017-01-04:
Thanks, I've spun up a 32bit version of Ubuntu and it seems to be compiling OK now.

Was just hoping you could clarify the final part of my original post while I have your attention, what's the correct options for make to build for the Due/gShield?  I've spent ages looking through the wiki and I think I know but I'm not 100% sure, and long the same lines, do any of the release work with the due/gShield?

Thanks,

Stephen.

giseburt, 2017-01-04:
Ok good. There are also 32-bit compatibility libraries for 64-bit Linux, so
it'll work in a 64-bit Linux.

The compile options would be something like:

make BOARD=gShield SETTINGS_FILE="..."

The BOARD=gShield part is case sensitive. The settings file would be
specific to your machine or one of the defaults found in settings/. For
example: SETTINGS_FILE="settings_othermill.h"

Does that help?
-Rob

perfectphase, 2017-01-04:
Thanks, that's a help. Just for context, I have a couple of TinyGv8 boards and have my own app that feeds gcode to it, and am looking to have a play with g2 to see if the changes to the comms and alarms help with the issues I'm having.

Cheers,

Stephen.

ril3y, 2017-01-04:
perfectPhase,

Have you posted your code?  We are always interested in new stuff for TinyG
and G2.

Thanks!

wedges, 2017-01-04:
@perfectphase , 
You just have to add some i386 lib to your x64 linux;
On ubuntu: `sudo apt-get install -y libc6-i386 lib32stdc++6 lib32gcc1 lib32ncurses5 lib32z1`

But no way to make it work with Windows 10+Bash...

adleal, 2017-02-10:
@perfectphase 

I've run into the same problem but since I'm compiling on Raspbian a 32-bit system I'm not sure why I'm getting the same error you were:

make: *** [build/TestV9-g2v9k/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.o] Error 126

giseburt, 2017-02-10:
@adleal You've run into another issue altogether. It appears we check for
Linux but not architecture, so it's downloading the Intel tools and you're
on an ARM processor.

Since the tools we use have been moved to ARM's web site, this has changed.
They don't have an ARM build (interestingly) and the Linux build is 64-bit
only now.
https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads

So some refactor is needed it appears.

adleal, 2017-02-10:
@giseburt 

Thanks for the clarification Rob... I wasn't even considering the difference in architecture. So I guess my best bet for now would be to move to a 32-bit Intel based Linux system and recompile there.

justinclift, 2018-04-12:
Hey all, this issue looks pretty old and it seems like it's solved.

Is it ok to close this one, or are there bits that still need stuff done? :smile:

justinclift, 2018-04-12:
Hmmm, it should work on 64-bit Linux "fairly easily" these days.  I say "fairly easily" because when I tried it a few weeks back on CentOS 7 64-bit it "just worked".

Hadn't actually realised until reading this issue that there had been 32-bit vs 64-bit problems in the past, nor architecture issues to work through (eg compiling on ARM vs Intel host).

But, that's just one fairly-new-to-the-project persons viewpoint so far.  If you want to try compiling stuff again on a recent 64-bit Linux system that'd help verify things. :smile:

adleal, 2018-04-12:
Ok will see if I can try it next week.

justinclift, 2018-04-12:
Awesome, please do. :smile:

giseburt, 2018-04-12:
Some more info: The versions of the gcc toolchain we use (built by ARM) in
edge and master are only available in 32-bit for Linux. So for most Linux
machines running in Intel (or compatible) hardware you’ll need the 32-bit
compatibility installed. For many people they may have already installed
those, for others they’ll see this issue pop up.

Now for the near future branches like Dev-168 that we are wrapping up and
will pull soon, they use a newer version of the ARM toolchain (to get C++17
support, along with other bug fixes) which is now the opposite: they’re
only available for 64-bit on Intel Linux. For most people that will not be
an issue. For those on 32-bit intel Linux there’s not really a solution. :-/

This isn’t really something we have control over, unfortunately. We are
using the gcc and related build tools provided by ARM, and they stopped
building them for 32-but Linux.

-Rob

justinclift, 2018-04-12:
k.  That sounds like we'll be requiring 64-bit Linux (at least on Intel) in the near-ish future.

When that happens, we should update our docs to mention it, just to keep things clear for people. :smile:

adleal, 2018-04-25:
> @adleal <https://github.com/adleal> Reminder ping. 😄

Sorry for the delayed response but I had too many things on my plate last
week. Today I was trying to install a 64 bit Unix system on the same Linux
box I was running the 32 bit system that finally allowed me to compile
everything... but unfortunately I ran into some hardware issue with
overheating (which I never had before)... so it looks like I won't be of
much help.

adleal, 2018-04-25:
Okay, I was able to get 64-bit Linus system running but when I went to compile I got this error:

make[1]: Leaving directory '/home/alex/g2/Motate/Tools'
Compiling c ../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.c
    -> build/TestV9-g2v9k/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.o 
bash: ../Motate/MotateProject/motate/../../Tools/linux/gcc-arm-none-eabi/bin/arm-none-eabi-gcc: No such file or directory
../Motate/MotateProject/motate/Motate.mk:475: recipe for target 'build/TestV9-g2v9k/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.o' failed
make: *** [build/TestV9-g2v9k/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.o] Error 127

justinclift, 2018-04-25:
Damn.  That looks like it's not downloading the right bits you for some reason.  With the 64-bit Linux you got running, which distro and which architecture? :smile:

adleal, 2018-04-25:
Okay after rereading the thread carefully I installed the 32-bit Linux compatibility using:

**sudo apt-get install lib32ncurses5 lib32z1**

and was then able to compile successfully:

_Exporting symbols bin/TestV9-g2v9k/g2core.elf.txt 
Making binary bin/TestV9-g2v9k/g2core.bin 
--- SIZE INFO ---
   text	   data	    bss	    dec	    hex	filename
 162896	      0	  35024	 197920	  30520	bin/TestV9-g2v9k/g2core.elf
Build 100.26 100.26-10-g1ff7_

justinclift, 2018-04-26:
Cool! :smile: