synthetos/g2 #208: Line number

alessandrolecce, 2016-12-25:
In Text mode ( I don't know in JSON) gcode with N code  (line numeber) are non report.... just line 0 or -0 or -2
report on serial is:
Line Number: 0
if there is N code in gcode, or nothing if there is not N code in gcode.

I think the problem is in 
gcode_parser.cpp line 623

but I just report to explorer what's happen...I'm not a good C++ programmer :(

alessandrolecce, 2016-12-25:
in Json mode report line is ok!

alessandrolecce, 2016-12-25:
ok...the problem is in text_parser.h line: 58 
There is a bug on new implementation.

aldenhart, 2016-12-25:
Thanks for the heads up. It looks like the printf() format for line numbers in incorrect. We'll fix this in a future update.

aldenhart, 2017-01-03:
Fixed in PR #211 - build 100.17

closed 2017-01-03 by aldenhart