synthetos/g2 #209: Random GCode Lines Skipped & Stopping in the middle of a Cycle.

antbarney, 2016-12-26:
Well working on a Project over the holiday to make some engraved discs. I have found that Some times entire lines of Gcode will be skipped or the Cycle will just stop mid process. I am using ChiliPepper to stream the GCode and build 100.15 (Just did the Build this morning after pulling the latest updates). 

Here is one of the files I have had the most issues with. It will complete some times but most times it will stop in the middle of the cycle or miss sections in the middle. Just checking if this is related to an of the known issues being worked.

[Katelin_Front_0001.txt](https://github.com/synthetos/g2/files/673341/Katelin_Front_0001.txt)

johnlauer, 2016-12-26:
My guess is cuz your lines and or decimals are too long.

Some of your gcode…

(Using default header. To add your own header create file "header" in the output dir.)

G02 X16.167383 Y19.391749 Z-1.000000 I-1.098619 J-0.120748 F300.000000

antbarney, 2016-12-27:
I tried reducing that to 4 decimal places before but figured I would give it another go since you also suggested it. Also removed all Spaces and comments. Still have it stall from time to time. 

[Katelin_Test_4Digit_0002.txt](https://github.com/synthetos/g2/files/673416/Katelin_Test_4Digit_0002.txt)

johnlauer, 2016-12-27:
Just ran it on tinyg v8. Ran like a champ.

When you say stall, what do you mean? Do you have to hit resume? Or does it just pause and resume on it’s own, cuz if so you’re not uploading lines fast enough.

johnlauer, 2016-12-27:
Made a video of the perfect run

https://www.youtube.com/watch?v=3T4jllq-JZE

antbarney, 2016-12-27:
First I should have mentioned this is on a Due running the latest g2core Edge Build with custom wiring running external drivers. 

When I say stall I mean the cycle stops with a number of lines in the SPJS buffer and hitting resume can some times get it to start again but that is normally only for the next line. I can recover if I wipe the buffer and restart, I think it is some new bug in the edge builds as I did not notice this before. Also removed my Limit switches to make sure it was not accidentally being triggered.

I have also tried now with decimal places down to 3 
[Katelin_Test_3Digit.txt](https://github.com/synthetos/g2/files/673445/Katelin_Test_3Digit.txt)

Screenshot of what ChiliPeppr Looks like after it hangs. 
[image]

System info.
{"r":{"sys":{"fb":100.15,"fbs":"100.15-7-g9d372","fbc":"settings_cbeam.h","fv":0.99,"hp":3,"hv":0,"id":"0084-d639-18c6-9ce","jt":1.25,"ct":0.01,"sl":0,"lim":1,"saf":0,"m48e":1,"mfoe":0,"mfo":1,"mtoe":0,"mto":1,"mt":10,"spep":1,"spdp":0,"spph":1,"spdw":1,"ssoe":0,"sso":1,"cofp":1,"comp":1,"coph":1,"tv":1,"ej":1,"jv":2,"qv":0,"sv":1,"si":250,"gpl":0,"gun":1,"gco":1,"gpa":2,"gdi":0}},"f":[1,0,3]}

johnlauer, 2016-12-27:
Notice in your sr responses it skips from line 78 to line 83. That looks whacky to me. It should have an sr for 79, 80, 81, 82, etc. without those sr’s, which is what spjs looks for to know the buffer flow can open back up again, you are dead in the water. When you hit resume you force the buffer back open until the same bug happens again. This is synthetos’s bug from what I can tell.

[test]<https://cloud.githubusercontent.com/assets/17607285/21488192/66b08d16-cb91-11e6-8a79-37644b01a705.png>

antbarney, 2016-12-27:
Ya, I was pretty sure it was a bug in the TinyG2 Code. I also noticed that some of the response messages due contain errors after forcing the resume so I think some of the GCode is getting garbled/not fully buffered properly. As these sames lines will some times work.

some examples:
{"r":{},"f":[1,102,50]}
{"sr":{"stat":3}}
{"r":{},"f":[1,154,37]}

giseburt, 2016-12-27:
The SR can skip lines. You can't rely on SR to report every line. If you
need to see every line be accepted, you have to set the JV to 4 so it'll
report the line number in n.

Explanation: SR will only report the current state when the report was
generated. Status reports are intentionally throttled and spaced out. They
are generally returned relatively evenly timed during a job, but if there
are a large amount of very quick moves (which requires extra planning time)
they may be throttled more than usual, and even more lines will be
"skipped."

All that said, I thought CP relied on the {r:...}, not the SR. Is this
correct?

-Rob

https://github.com/synthetos/g2/files/673445/Katelin_Test_3Digit.txt>

https://cloud.githubusercontent.com/assets/17607285/21488192/66b08d16-cb91-11e6-8a79-37644b01a705.png

System info.

{"r":{"sys":{"fb":100.15,"fbs":"100.15-7-g9d372","fbc":"settings_cbeam.h","fv":0.99,"hp":3,"hv":0,"id":"0084-d639-18c6-9ce","jt":1.25,"ct":0.01,"sl":0,"lim":1,"saf":0,"m48e":1,"mfoe":0,"mfo":1,"mtoe":0,"mto":1,"mt":10,"spep":1,"spdp":0,"spph":1,"spdw":1,"ssoe":0,"sso":1,"cofp":1,"comp":1,"coph":1,"tv":1,"ej":1,"jv":2,"qv":0,"sv":1,"si":250,"gpl":0,"gun":1,"gco":1,"gpa":2,"gdi":0}},"f":[1,0,3]}

—

https://github.com/synthetos/g2/issues/209#issuecomment-269255942>, or mute
the thread<

<https://github.com/synthetos/g2/issues/209#issuecomment-269256383>, or mute
the thread

johnlauer, 2016-12-27:
Oh, sorry, I meant TinyG buffer in spjs needs an r: response to know when to release the buffer, not an sr

giseburt, 2016-12-27:
John: Ok. That's what I thought. But now anyone perusing this issue will
see why you can't rely on SR for each line. ;-)

Other than some fairly long lines I don't see anything that stands out as
an obvious problem. I'm on my phone now so I can't tell a lot. We'll look
into it and see what we can find.

-Rob

—

<https://github.com/synthetos/g2/issues/209#issuecomment-269257189>, or mute
the thread

antbarney, 2016-12-27:
Just to add some more info flashed back to the Master Branch build and ran this perfectly at least 3 times without any issues. Wanted to be sure to rule out anything in my build.

[fb]  firmware build             89.03
[fbs] firmware build "                           build"
[fv]  firmware version            0.98
[cv]  configuration version       7.00
[hp]  hardware platform           3.00
[hv]  hardware version            0.00
[id]  TinyG ID       0213-0215-d403

aldenhart, 2016-12-30:
@artbarney -  Can you please post your settings_cbeam.h file?

antbarney, 2016-12-30:
[settings_cbeam_edge.txt](https://github.com/synthetos/g2/files/678883/settings_cbeam_edge.txt)
[settings_cbeam_master.txt](https://github.com/synthetos/g2/files/678884/settings_cbeam_master.txt)

Here is the settings file I used to do the builds from both the edge and master branch

stg, 2016-12-31:
If you are not running a mill/router, or you are getting this error while testing without the spindle motor running you can stop reading - otherwise I'd like to suggest EMC and static electricity as possible causes.

I've had this error many, many times early on. The controller would sometimes skip a random number of g-code lines and force the tool straight through the workpiece. Sometimes it would stop dead in its tracks. After a stop it could sometimes be recovered, sometimes not. The USB connection would sometimes break and require a reconnect before accepting anything at all. It all sounds too similar to not mention it - sorry for running my mouth if it does not apply, but here goes:

I was using a standard chinese spindle @1.5kW/24krpm. These motors put out an absurd amount of EMC especially when starting/stopping. More than enough to glitch out any nearby unprotected system and wreak havoc on any high impedance end-stops, touch-plate, other sensors. Cutting certain materials such as MDF and many plastics also cause build up of static electricity on the motor and Z axis as many machines are electrically insulated between Z and X as well as X and Y axes. You may notice static shocks when touching the motor during/after runs.

Both phenomena can enter and travel along cables to the controller - in my case also an Arduino Due which appears to be fairly sensitive.

To solve the problem I had to encapsulate the Arduino in a metal case and run external cables from a proper ground connection to the Dues new case, the CNC machine bed, and one point on each axis to make sure any static build up would discharge immediately. The last bit I needed for absolute stability was some ferrites on the cabling and a shorter, better shielded usb cable.

As for the sensors, one needs to make sure every signal in or near the machine is fairly low-impedance or heavily filtered - not true for out-of-the-box chinese machines.

Every time I accidentally disconnect ground (a gator clip on the nearest copper piped radiator) the problem comes back so by now I'd say it's 100% verified.

antbarney, 2016-12-31:
Yes this is testing with the spindle off. And as stated only seems to affect the edge builds. Master branch builds look to work fine.

cmaccioli, 2017-01-02:
I am having a similar issue. Just got the system set up last weekend and have been able to run small gcode samples, but when I try to run a large file, the systems hangs.  Sometimes it starts back up, but if I flush the planner buffer it starts running again.  It almost always stops when the planner hits 48. I just pulled the latest 100.17 and it is still happening. Running in chillipepper, with tinyg buffer (tried tinyg2 as well).  I can get it to run more lines by changing gcode sender to 100 pre-upload, no multiline and delay to 50 ms.  
This happens running a laser or spindle, I have ferrite beads on all low level signals, ground wire, etc.  Running on Due with no shield, 3.3v stepper to Intelligent motions IM804 stepper drivers.
Running JSON ver. 1.92 on windows 8 (x64).

aldenhart, 2017-01-03:
@artbarney - we have not been able to make CP / 100.xx fail with the Kaitlin files. Do you have a longer file that does this? Also, what are you running as a host - Windows 10, OSX, rPi...?

@cmaccioli - can you post a file that fails for you?

antbarney, 2017-01-03:
Sure Here is a bigger one I just tested, Fails at random spots also. All tests have been using ChiliPeppr SPJS 1.92, I have tried from Win10,rPi3,Ubuntu but all where with the SPJS 1.92
 
[Table.txt](https://github.com/synthetos/g2/files/681619/Table.txt)

aldenhart, 2017-01-03:
@antbarney - Question for you - are you able to  run the default Gcode file that opens up with the CP TinyG workspace? I'm running this fine - before trying your files. If you cannot run it without issue it points to something in the toolchain - more than likely. I'm running on OSX 10.11, CP 1.92, set to 100 line pre-load. All other CP parameters are defaults.

cmaccioli, 2017-01-03:
@aldenhart - Ill upload a file tonight when I get home.  I'll try the default Gcode (CP logo) when I get home as well. I'll also try some of the files posted by @antbarney. 

- I was using SPJS 1.88 with same issues, but updated yesterday to 1.92 to see if it made a difference (it did not).

Using Atmel Studio 7 to compile. Did have an error initially and it wouldn't compile, but had error in my settings file regarding ABSOLUTE_DISTANCE_MODE  not defined, but addressed that and compiled ok. 

Question: The planner status in CP, was typically at 48 when it stopped, does this mean the planner was overloaded or their was nothing planned?  After several tries, I left the machine stay stopped for 10-30 seconds and sometimes it would start back up and the planner would drop below 48.  
I was also able to continually press the ~ (cycl start) in CP and the machine would basically step through the program, sometimes running 1 line of code, sometimes running 2-3 lines.

Sorry of I am giving too much or impertinent info. I am new with tinyg2 and am learning as I go.  I am reading thru the code and wiki and gaining a better understanding, but it will take a while.

johnlauer, 2017-01-03:
Sounds like same issue that about 3 other folks have mentioned. I think r:{}’s are disappearing coming back from TinyG and thus the buffer locks.

antbarney, 2017-01-03:
I will try with the standard CP file when I get home today and report.

ewidance, 2017-01-03:
I got exactly the same problem, using OSX / Firefox / SPJS 1.92 / Edge. I got no spindle, only a laser. I tried with many files, espacially one generated with arduino Uno 2 Top layer. It's stops very early in file. When i hit "%" (Queue flush) it starts again then stops.

cmaccioli, 2017-01-03:
Was able to run the CP logo (mm) without issue.
Tried running my file and it stopped at various lines.
Tried running table.txt file from @antbarney and stopped at line 485.
Each time I tried to run it seems like it would stop earlier until I stopped JSON and restarted chrome.
[catcher_0006.txt](https://github.com/synthetos/g2/files/683534/catcher_0006.txt)

Also tried new USB cable just in case.
Could there be a driver issue? I know at first, I had to fudge the Tinyg.inf file because the ID of the port was different.

One time after it stopped, I turned off the filter in serial port console and hit the ~ button and here is what I got back : (it stopped at line1647, I hit ~ and it ran until line1664)
"~

{"qr":""}
{"Lbs":199}
{"r":{"n":1648},"f":[1,0,41]}
{"Lbs":183}
{"r":{"n":1649},"f":[1,0,27]}
{"qr":46}
{"Lbs":182}
{"r":{"n":1650},"f":[1,0,27]}
{"qr":45}
{"Lbs":165}
{"r":{"n":1651},"f":[1,0,43]}
{"Lbs":163}
{"r":{"n":1652},"f":[1,0,27]}
{"qr":43}
{"Lbs":136}
{"r":{"n":1653},"f":[1,0,12]}
{"qr":42}
{"Lbs":109}
{"r":{"n":1654},"f":[1,0,12]}
{"qr":40}
{"Lbs":66}
{"r":{"n":1655},"f":[1,0,15]}
{"Lbs":169}
{"r":{"n":1656},"f":[1,0,27]}
{"qr":39}
{"Lbs":157}
{"r":{"n":1657},"f":[1,0,12]}
{"qr":38}
{"Lbs":188}
{"r":{"n":1658},"f":[1,0,15]}
{"qr":36}
{"Lbs":173}
{"r":{"n":1659},"f":[1,0,12]}
{"qr":35}
{"Lbs":146}
{"r":{"n":1660},"f":[1,0,21]}
{"Lbs":134}
{"r":{"n":1661},"f":[1,0,43]}
{"Lbs":199}
{"r":{"n":1662},"f":[1,0,43]}
{"Lbs":187}
{"r":{"n":1663},"f":[1,0,37]}
{"Lbs":210}
{"r":{"n":1664},"f":[1,0,43]}
{"sr":{"line":1648,"posx":19.783,"posy":6.313,"vel":145.52,"stat":5,"momo":3,"mpox":214.783,"mpoy":181.313}}
{"qr":32}
{"sr":{"line":1651,"posx":19.971,"posy":6.597,"vel":50.48,"mpox":214.971,"mpoy":181.597}}
{"qr":35}
{"sr":{"line":1656,"posx":19.248,"posy":6.235,"vel":802.97,"feed":3000,"momo":1,"mpox":214.248,"mpoy":181.235}}
{"sr":{"posx":14.014,"posy":3.531,"vel":1563.36,"mpox":209.014,"mpoy":178.531}}
{"sr":{"posx":10.843,"posy":1.893,"vel":93.5,"mpox":205.843,"mpoy":176.893}}
{"sr":{"line":1662,"posx":11.027,"posy":2.17,"vel":198.59,"feed":500,"momo":3,"mpox":206.027,"mpoy":177.17}}
{"qr":46}
{"qr":48}
{"sr":{"line":1664,"posx":10.774,"posy":2.737,"vel":0,"stat":3,"mpox":205.774,"mpoy":177.737}}
{"sr":{"stat":3}}

aldenhart, 2017-01-04:
@cmaccioli - Thanks for the detailed dump. I can't find a smoking gun, but here's what we can learn from it. 

- The "Lbs" commands are issued to SPJS only and are not sent to the g2 board. They can be ignored. (John, please correct me if I'm wrong about this).

- The "qr" commands are a response from g2 saying how many free planner buffers it has - up to a maximum of 48. If the planner is full this number is usually 3 (it keeps some free for odd commands). It returns 48 when the buffer is empty. 

- The fact that the buffers were hovering in the mid-levels (not full, at qr:3) indicates that there are very short Gcode lines being sent, and the buffer and play-out are throttling. g2 throttles the top speed of the Gcode lines if the sender / USB system cannot keep up with the rate the lines want to play out. The effect is a planner buffer that oscillates around the mid-levels as you see here. 

So what this tells us is that the buffer was almost empty at line 1648, then filled up some, then stopped receiving any new input after line 1664, for which it received a response ( {r:...} ). The {r} was returned, and a status report {sr:...} later, once it was executed. It looks like line 1664 was the last line sent t the board, as this coincides with the buffer emptying {qr:48}.

The question is - did SPJS ever send line 1665? If it did, is this logged, and was the {r} returned or not? from the look of this dump it appears that it was never sent, but I don't know the details of the logging.

Some potential diagnostics:

- Once stalled, query the g2 board with a {"line":null} command. It will return the last line number it received. See if this matches the dump. Was it 1664 or 1665 (or later)?

- Have the CP dump flag any non-zero status responses --> "f":[1,0,43] is right, "f":[1,100,43] is an example of an error code whose line cannot be parsed. ANY non-zero status response is an error, and should be flagged somehow. I don;t see any here.

cmaccioli, 2017-01-04:
@aldenhart -
OK - for the buffer - The machine actually stopped at 1577 and I hit the ~ several times, then thought to turn off the filtering when it stopped at 1647 so I could see the status and saved the status info after that.
I am pretty sure that CP had sent all of the lines of code already.

Is there a way of outputting the status info to a text file directly while running?
Sorry if this is a stupid question, but how do I have CP dump flag any non-zero status responses?

I am connected directly to the USB port of my laptop, no Hub. 
Are we all seeing a USB port speed issue?
FYI, in windows device manager, the port shows up as "TinyG v2 (Data Channel) (Com 11)" would that be correct?
I will try to get more info now.

cmaccioli, 2017-01-04:
So I tried running my file and it stopped at line 524, but CP kept sending info and the serial monitor, truncated the data.
Note that this is the earliest Line # I have seen it stop at.
Tried running the  {"line":null}, but got nothing in response.
I stopped GCODE, then flushed the planner and got the following:
%

{"qr":""}
{"er":{"fb":100.17,"st":157,"msg":"Arc specification error - missing offset(s) - arc offsets missing or zero"}}
{"Lbs":0}
{"r":{"n":525},"f":[1,157,26]}
{"r":{"qr":48},"f":[1,0,11]}
{"sr":{"line":525,"stat":2}}
  {"line":null}
{"Lbs":0}
{"r":{"line":525},"f":[1,0,15]}

aldenhart, 2017-01-04:
The g2 board does not have memory for any logging. So all logging must be done at the host end, or on the USB wire. What I'd like to see from the log is:

1. Were there any non-zero status returns? Indicating either a corrupted transmission or some garbled transmission.

1. Were there any corrupted return lines that did not pass the JSON parser? Indicating a corrupted response. If the board is truncating a response line it won't pass JSON parsing (99.999% of the time!)

1. What was the last line number sent form SPJS, and what was the last line number received in an {r:}?

1. Did the board receive any lines where the line numbers did not monotonically increase by 1? Indicating a dropped line from SPJS.

Can you think of any other tests?

Be aware that for #3 the {r} test if the R "was not received" by the logger it does not necessarily mean that it was not sent by the board. That could be the case, or it could have been not picked up by the receiver, somehow. It's unlikely that the USB channel failed, as that would have most likely caused one of the corruption cases.

What we may need to do is get a USB sniffer to see what was actually sent in both directions. Wireshark can do this on OSX / Linux / Windows with a USBpcap so it might be necessary to configure a USB capture to really look under the covers.

I don't see the 100 builds completely missing a transmission - although this is possible - but it would not surprise me if the TX timing is somewhat different than the earlier releases.

aldenhart, 2017-01-04:
To your comment - If there is an arc specification error - or any other Gcode error - the job will stop.

If it made it past this point before it indicates that something was supported, as not all the modal "requirements" for an arc were satisfied.

What is the Gcode for line 525 and the lines leading up to it?

antbarney, 2017-01-04:
Is this a screw drive machine with multistart screws? - This is a Openbuilds Cbeam machine. ACME screws on all axis. 

Also, power at 1.0 is high. What stepper drivers are you driving?  Are you getting driver overheats? - I am using external TB6600 drivers and since I am only running Nema 23's at 2.8amp they are not even getting above Room temp. and with external drivers I don't think the Power setting even takes effect I could most likely just set those to 0 and be ok.

antbarney, 2017-01-04:
I also just tried the standard Gcode file from Chilipeppr and it also failed. 

Host rPi3 SPJS 1.92 Presend 100 lines rest default.

Here is the info I got after I flushed the buffer. 

N279 G3 X-11.7011 Y4.72 I-2.2027 J-2.9617

 {"line":null}
%

{"qr":""}
{"er":{"fb":100.17,"st":157,"msg":"Arc specification error - missing offset(s) - arc offsets missing or zero"}}
{"r":{},"f":[1,157,30]}
{"r":{"qr":48},"f":[1,0,11]}
{"sr":{"line":279,"stat":2}}

aldenhart, 2017-01-04:
Right. The power level is irrelevant for external drivers. You can turn it off.

Question - do the Toshiba drivers take an active LOW enable signal? Or did you invert the enable somehow?

antbarney, 2017-01-04:
I did not do any inverting manually. So I think they are Active HIGH. 
This is the one I used specifically. 

https://www.amazon.com/gp/product/B016ZJS1FA/ref=oh_aui_search_detailpage?ie=UTF8&psc=1

cmaccioli, 2017-01-04:
I am running THK Linear actuator (ball screw with guides) using NEMA 23 with IM804 drivers runnning 16 microsteps @ 3.2 amp.  Inverted enable signal.

Just ran again and ran the JSON server in verbose mode out to text file and am attaching it.
I preloaded 10,000 lines (full program = 2980 lines).

[log.txt](https://github.com/synthetos/g2/files/683713/log.txt)

antbarney, 2017-01-04:
Also did a Verbose log of it failing with the standard CP logo file.

[Verbose_SPJS_Output.txt](https://github.com/synthetos/g2/files/683720/Verbose_SPJS_Output.txt)

aldenhart, 2017-01-04:
Unfortunately the verbose SPJS log doesn't help me much. It has all manner of things mixed in an it's hard to find what the actual back and forth is masked. What I'm trying to get at is something like:

>>> Sent:  (stuff sent to board)
<<< Recd: (stuff received from board)

Ideally w/ timestamps. We've built this for some of our other integrations (e.g. Shopbot) and it's a good way to get the actual data exchanges.

On the v8 we tap the serial lines and feed them to a Logic protocol analyzer to see this data. But the g2 stuff is pure USB, so we don't have those signals.

johnlauer, 2017-01-04:
It would be cool if somebody added a command line flag to SPJS like -log and then added 2 new lines of code in the buffer that ONLY wrote outgoing and incoming lines. It wouldn’t be super hard.

aldenhart, 2017-01-04:
I just ran the CP Logo Gcode (mm) with no issues. Is this the file you are running that is failing?

cmaccioli, 2017-01-04:
@aldenhart - I was able to run that file without issue.  Not sure which file @antbarney was running though.

cmaccioli, 2017-01-04:
@aldenhart Downloading winshark. never used it, so are there any specific instructions to get the data you might need?

giseburt, 2017-01-04:
This part looks interesting:

2017/01/04 01:35:02 serial.go:357: Consuming sendBuffered queue. ok:true,
d:N584 M5
, id:g584
2017/01/04 01:35:02 serial.go:357: Consuming sendBuffered queue. ok:true,
d:N585 M30
, id:g585
2017/01/04 01:35:02 serial.go:368: Hit default in select clause
2017/01/04 01:35:02 serial.go:371: Done consuming sendBuffered cmds. ctr:352
2017/01/04 01:35:02 bufferflow_tinygg2.go:754: Lock being released in TinyG
buffer
2017/01/04 01:35:02 bufferflow_tinygg2.go:867: Just sent release to b.sem
so we will not block the sending to serial port anymore.
2017/01/04 01:35:02 serial.go:383: itemsInBuffer:1
2017/01/04 01:35:02 bufferflow_tinygg2.go:246: Done blocking cuz got b.sem
semaphore release. ok:true, unblockType:2
2017/01/04 01:35:02 bufferflow_tinygg2.go:251: This was an unblock of type
2, which means we're being asked to wipe internal buffer. so return false.
2017/01/04 01:35:02 serialport.go:246: We got back from BlockUntilReady()
but apparently we must cancel this cmd
2017/01/04 01:35:02 serial.go:401: We need to unpause our internal buffer.

cmaccioli, 2017-01-04:
Pretty sure his machine stopped a ways above that at 
2017/01/04 01:33:43 bufferflow_tinygg2.go:309: < {"sr":{"posx":-1.556,"posy":0.128,"vel":0,"stat":3,"mpox":-1.556,"mpoy":0.128,"cycs":0}}
2017/01/04 01:33:53 bufferflow_tinygg2.go:309: < {"sr":{"stat":3}}.

antbarney, 2017-01-04:
Correct, the machine stopped before that, and I had to flush the buffer (send "%") before I could get {"line":null} to return.

in my log file I ran the CP Logo file from the Chilipeppr Page (mm version) I have ran in a few times and have had it complete successfully at least once. And failure points in the other cases where in different spots.

On 1/3/2017 6:00 PM, cmaccioli wrote:

Pretty sure his machine stopped a ways above that at
2017/01/04 01:33:43 bufferflow_tinygg2.go:309: < {"sr":{"posx":-1.556,"posy":0.128,"vel":0,"stat":3,"mpox":-1.556,"mpoy":0.128,"cycs":0}}
2017/01/04 01:33:53 bufferflow_tinygg2.go:309: < {"sr":{"stat":3}}.

aldenhart, 2017-01-04:
@antbarney - I know you said this before, but this thread is getting long. What machines are you running? This all sounds like comms problems on specific machines. I am having no troubles  running this from a clean OSX 10.11. (Otherwise we would not have pushed to edge yet.)

antbarney, 2017-01-04:
As the Host for SPJS I am running it from a dedicated Raspberry Pi 3 with SPJS 1.92, but I have also tried from a Fresh Win 10 Machine, as well as Ubuntu 16.04.1. 

G2 is running on a Arduino Due with custom wiring for the drivers.

aldenhart, 2017-01-04:
Did you get better results on W10 or Ubuntu? The rPi was known to be underpowered. I'm not sure about the 3's, though.

antbarney, 2017-01-04:
No, had the same issues on Ubuntu and W10. The Pi 3 seems to handle it pretty well have been using it for a while without issue.

aldenhart, 2017-01-04:
Do all the Kaitlins fail about the same? I think I should run them next. Until I can reproduce the issue I'm not sure what I can do. So far I've been working up to it and everything I've run completes without issues

antbarney, 2017-01-04:
They never fail in the same spots from what I can tell but they do fail regularly. That's what seems odd to me is sometimes it will successfully complete, but it will fail pretty regularly for me anyway.

giseburt, 2017-01-04:
In the log, starting at 2017/01/04 01:33:33 (I can't get my phone to copy
it) SPJS says it's pausing, then 4 {r...} responses go by. It never I
paused as far as I can tell. That looks like it's the root of the problem
in there.

antbarney, 2017-01-04:
[CP_Logo_Fail.txt](https://github.com/synthetos/g2/files/683823/CP_Logo_Fail.txt)
I know you said that the SPJS logs where not much help but wanted to just add another set this time from a W10 Host. 

I have also removed all Limit switches from my setup just to make sure there is not some glitching going on there (which I have checked many times before this also and always confirmed via $in that the input states where correct.)

antbarney, 2017-01-04:
I also just had it fail basically at the start of the CP Logo file. and looks like the response is an error 154 (STAT_ARC_HAS_IMPOSSIBLE_CENTER_POINT) which from what I can was for line N37 G2 F800.0 X31.7601 Y-2.7342 I-5.0775 J0.7343 which is valid if at the proper starting point of X31.8196 Y-2.4008 Z0 and will run fine when I manually execute. Could the g2 Buffer or planner be missing actions and not be in the correct location when some of the lines get executed causing it to enter an error state? Sorry I am not all to much help really a noob when it comes to stuff like this.

2017/01/03 18:44:41 bufferflow_tinygg2.go:309: < {"sr":{"line":16,"posx":38.563,"posy":-4.195,"posz":0,"vel":800,"feed":800,"momo":2,"mpox":38.563,"mpoy":-4.195,"mpoz":0}}
2017/01/03 18:44:42 bufferflow_tinygg2.go:309: < {"sr":{"line":20,"posx":35.934,"posy":-3.092,"mpox":35.928,"mpoy":-3.073}}
2017/01/03 18:44:42 bufferflow_tinygg2.go:309: < {"sr":{"line":23,"posx":36.155,"posy":0.141,"mpox":36.155,"mpoy":0.141}}
2017/01/03 18:44:42 hub.go:122: sendjson {"P":"COM7","Data":[{"D":"N38 G2 X31.2841 Y-3.7614 I-2.3626 J0.471\n","Id":"g38"},{"D":"N39 G2 X30.6633 Y-4.2692 I-1.5584 J1.2717\n","Id":"g39"},{"D":"N40 G2 X29.8824 Y-4.452 I-0.7659 J1.5128\n","Id":"g40"},{"D":"N41 G2 X29.1178 Y-4.2757 I-0.0154 J1.6797\n","Id":"g41"},{"D":"N42 G2 X28.5081 Y-3.7819 I0.9002 J1.7348\n","Id":"g42"},{"D":"N43 G2 X28.0487 Y-2.7942 I1.8061 J1.4408\n","Id":"g43"},{"D":"N44 G2 X27.9406 Y-1.7102 I4.813 J1.0272\n","Id":"g44"},{"D":"N45 G2 X28.0605 Y-0.5925 I4.765 J0.0543\n","Id":"g45"},{"D":"N46 G2 X28.5423 Y0.4231 I2.4581 J-0.5442\n","Id":"g46"},{"D":"N47 G2 X29.154 Y0.9543 I1.7404 J-1.3862\n","Id":"g47"}]}
2017/01/03 18:44:42 serial.go:252: Done with writeJson method
2017/01/03 18:44:42 serialport.go:233: Got p.sendBuffered. data:N38 G2 X31.2841 Y-3.7614 I-2.3626 J0.471\n, id:g38, pause:0
2017/01/03 18:44:42 bufferflow_tinygg2.go:192: BlockUntilReady(cmd:N38 G2 X31.2841 Y-3.7614 I-2.3626 J0.471
, id:g38) start
2017/01/03 18:44:42 bufferflow_tinygg2.go:225: New line length: 41, buffer size increased to:90
2017/01/03 18:44:42 serialport.go:233: Got p.sendBuffered. data:N39 G2 X30.6633 Y-4.2692 I-1.5584 J1.2717\n, id:g39, pause:0
2017/01/03 18:44:42 bufferflow_tinygg2.go:192: BlockUntilReady(cmd:N39 G2 X30.6633 Y-4.2692 I-1.5584 J1.2717
, id:g39) start
2017/01/03 18:44:42 bufferflow_tinygg2.go:225: New line length: 42, buffer size increased to:132
2017/01/03 18:44:42 bufferflow_tinygg2.go:232: It looks like the buffer is over the allowed size, so we are going to pause. Then when some incoming responses come in a check will occur to see if there's room to send this command. Pausing...
2017/01/03 18:44:42 bufferflow_tinygg2.go:243: Blocking on b.sem until told from OnIncomingData to go
2017/01/03 18:44:42 serialport.go:268: Got p.sendNoBuf. id:g38, pause:0, data:N38 G2 X31.2841 Y-3.7614 I-2.3626 J0.471\n
2017/01/03 18:44:42 serialport.go:275: Items In SPJS Queue List:9
2017/01/03 18:44:42 bufferflow_tinygg2.go:309: < {"r":{},"f":[1,154,41]}
2017/01/03 18:44:42 bufferflow_tinygg2.go:345: Buffer decreased to itemCnt:2, lenOfBuf:83
2017/01/03 18:44:42 bufferflow_tinygg2.go:867: Just sent release to b.sem so we will not block the sending to serial port anymore.
2017/01/03 18:44:42 bufferflow_tinygg2.go:246: Done blocking cuz got b.sem semaphore release. ok:true, unblockType:1
2017/01/03 18:44:42 serialport.go:233: Got p.sendBuffered. data:N40 G2 X29.8824 Y-4.452 I-0.7659 J1.5128\n, id:g40, pause:0
2017/01/03 18:44:42 bufferflow_tinygg2.go:192: BlockUntilReady(cmd:N40 G2 X29.8824 Y-4.452 I-0.7659 J1.5128
, id:g40) start
2017/01/03 18:44:42 bufferflow_tinygg2.go:225: New line length: 41, buffer size increased to:124
2017/01/03 18:44:42 bufferflow_tinygg2.go:232: It looks like the buffer is over the allowed size, so we are going to pause. Then when some incoming responses come in a check will occur to see if there's room to send this command. Pausing...
2017/01/03 18:44:42 bufferflow_tinygg2.go:243: Blocking on b.sem until told from OnIncomingData to go
2017/01/03 18:44:42 serialport.go:268: Got p.sendNoBuf. id:g39, pause:0, data:N39 G2 X30.6633 Y-4.2692 I-1.5584 J1.2717\n
2017/01/03 18:44:42 serialport.go:275: Items In SPJS Queue List:8
2017/01/03 18:44:42 bufferflow_tinygg2.go:309: < {"r":{},"f":[1,154,42]}
2017/01/03 18:44:42 bufferflow_tinygg2.go:345: Buffer decreased to itemCnt:2, lenOfBuf:83
2017/01/03 18:44:42 bufferflow_tinygg2.go:867: Just sent release to b.sem so we will not block the sending to serial port anymore.
2017/01/03 18:44:42 bufferflow_tinygg2.go:246: Done blocking cuz got b.sem semaphore release. ok:true, unblockType:1
2017/01/03 18:44:42 serialport.go:233: Got p.sendBuffered. data:N41 G2 X29.1178 Y-4.2757 I-0.0154 J1.6797\n, id:g41, pause:0
2017/01/03 18:44:42 bufferflow_tinygg2.go:192: BlockUntilReady(cmd:N41 G2 X29.1178 Y-4.2757 I-0.0154 J1.6797

cmaccioli, 2017-01-04:
I have wireshark installed. any tips on how to run it? filters, etc.
I have a capture, but there is a ton of data, not sure how to properly filter it.

cmaccioli, 2017-01-04:
Well, here is a wireshark log.  I filtered out interrupts and packets with data_len=0
[tinyg2_usb_log.zip](https://github.com/synthetos/g2/files/683872/tinyg2_usb_log.zip)
Zipped it, size was 274 mb

ril3y, 2017-01-04:
That file looks like it has nothing in it?  Perhaps I am missing
something?  Wireshark just captures the encapsulated or encoded USB
packets?  I dont see an easy way to see the data inside these packets.
Again perhaps I am missing something.

ril3y, 2017-01-04:
Nothing meaning nothing useful.  It has loads of USB packets.

cmaccioli, 2017-01-04:
Just out of curiosity, it sounds like @antbarney and myself are compiling on our side. Just wondering if we gave you our settings file and any other changes, if you could compile on your side to see if it might be something in how we are compiling. Not sure if it is a stupid question or not, but grabbing at straws.

On a side note I have run the catcher0006.nc file over 10 times and it has never stopped at the same spot.

cmaccioli, 2017-01-04:
@ril3y - I think you have to unzip and open with wireshark.
I had to filter out data that didn't look useful like packets with data length = 0 and packets that were interrupts. !(usb.transfer_type == 0x01) and !(usb.data_len == 0)
Then when I selected packets that had some data, I could see some info like"w	{"r":{"n":7},"f":[1,0,13]}"
[image]

ril3y, 2017-01-04:
Interesting. I did see that. I perhaps have an older version of Wireshark.
Will look again in the morning.

cmaccioli, 2017-01-04:
Just in case, here is my settings file and the gshield pinout.
I had to change pwm pin as my pin 13 didn't seem to work properly.
pin_number kSpindle_PwmPinNumber    = 9;
pin_number kSpindle_Pwm2PinNumber   = -1;

[settings_noshield.h.txt](https://github.com/synthetos/g2/files/683909/settings_noshield.h.txt)
[gShield-pinout.h.txt](https://github.com/synthetos/g2/files/683910/gShield-pinout.h.txt)

[81 more replies]

closed 2017-04-05 by giseburt