synthetos/g2 #212: kGRBL_CommonEnablePinNumber does not work anymore

ewidance, 2016-12-31:
Since 100 version, the grbl global enable pin (kGRBL_CommonEnablePinNumber) does not seems to  work anymore. 

I'm using Arduino Due + gShield defaut pinout. Pin is defined as is 
pin_number kGRBL_CommonEnablePinNumber = 8;

When connecting enable to one of those per axis enable signal, that works properly, with auto timeout desactivation.

aldenhart, 2017-01-04:
I think the best way to providing the common enable for the Due/Shield combo is simply to assign one of the motors' enable pins to the common enable pin, say motor1. Then use the motor power management mode MOTOR_POWERED_IN_CYCLE (2). That way whenever any motor is in cycle (active) the motor1 enable (now the common enable) will be active. 

This will require making these changes to the gShield-pinout.h file:

```
change:   pin_number kSocket1_EnablePinNumber  = 22;
to:            pin_number kSocket1_EnablePinNumber  = 8;

change:  pin_number kGRBL_CommonEnablePinNumber = 8;
to:           pin_number kGRBL_CommonEnablePinNumber = -1;
```

I'm not ins a position to test this right now, but if this is correct we can make the change "permanent" in the gShield configs and merge it into edge.

ewidance, 2017-01-04:
Nice workaroud, thank you. I'll test it on friday, since i'm not closed to the machine for the moment. 
However, any explanation about  CommonEnablePinNumber not managed anymore?

aldenhart, 2017-01-04:
Yes, we refactored the stepper motors somewhat to accommodate sub-classing them (see the hobbyservo Z axis, for example) and lost the global enable capability. To put it back in as-is requires some gyrations, and objects don't know global scope, only their own. In short, this looked like a better solution.

I will post a dev branch called dev-212-common-enable with these changes. Please let us know how the testing goes. If it's fine then we'll issue that as a pull request to the edge branch and bring it in.

ewidance, 2017-01-07:
OK, that works fine. For me ok for the pull request.

aldenhart, 2017-01-07:
OK. I'll close once we make the changes. Thx

FutureSharks, 2017-01-23:
I tested this just now on my Due/gShield and it now my motors timeout instead of being constantly on.

Would this solution work when motor1 is disabled?

justinclift, 2018-04-12:
Hmmm, does anyone know the state of this atm?  Seing some discussion that looks related (eg #178), but it's not clear if the fix has made it's way into anything yet. :wink: