synthetos/g2 #215: Kinematics : Corexy

ewidance, 2017-01-05:
Ive made some tests with CoreXY kinematics with a proto made with corexy.org files. I hacked kinematics.c 
```
static void _inverse_kinematics(const float travel[], float joint[]) {
    memcpy(joint, travel, sizeof(float) * AXES);  // just do a memcpy for Cartesian machines
    //	for (uint8_t i=0; i<AXES; i++) {
    //		joint[i] = travel[i];
    //	}
    #ifdef COREXY
    /* JPC: Corexy Kinematics test 
    deltaA=deltaX+deltaY
    deltaB=deltaX-deltaY
    */
    joint[0]=travel[0]+travel[1];
    joint[1]=travel[0]-travel[1];
    #endif
}
```

`
void kn_forward_kinematics(const float steps[], float travel[]) {
    float best_steps_per_unit[AXES];
...
...
    #ifdef COREXY
    /* JPC: Corexy Kinematics test 
    deltaX=1/2(deltaA+deltaB)
    deltaY=1/2(deltaA-deltaB)
    */
    travel[0]=0.5 * ( (steps[0] * st_cfg.mot[0].units_per_step) + (steps[1] * st_cfg.mot[1].units_per_step) );
    travel[1]=0.5 * ( (steps[0] * st_cfg.mot[0].units_per_step) - (steps[1] * st_cfg.mot[1].units_per_step) );
    #endif`
}

giseburt, 2017-01-05:
Interesting! How well did that work?

ewidance, 2017-01-05:
Very well, except i'm sticked with #209 and can't go further with laser engraving.. G2 stops :(

ewidance, 2017-01-07:
I've verified displacements with a caliper. Calibration is ok. Also verified homing and works fine.