synthetos/g2 #220: Hang on Command Spam [Due/gShield; 100.18-1-gf809-dirty]

hurr1k4ne, 2017-01-06:
My Due hangs when sending multiple commands to board. Single commands work, but all responses are slow (e.g. $ needs alsmost 2 sec.)
On stable 87 everything works, but after upgrade problem occurs. 

System: Windows 10, SPJS 1.92, Chilipeppr TinyG workspace
Board: Genuino Due, gShield V5
Firmware: 100.18-1-gf809-dirty (f809f31c5a82dabac0a4db66cfeeb84dfb95aebc)

Config: copied settings_default.h to settings_mymill.h and edited Motors and axis. Also exposed 2 Serial Ports
--------------------------------------------
 $
 {"ej":""}
[fb]  firmware build             100.18
[fbs] firmware build "100.18-1-gf809-dirty"
[fbc] firmware config "settings_mymill.h"
[fv]  firmware version            0.99
[hp]  hardware platform           3.00
[hv]  hardware version            0.00
[id]  g2core ID   0084-8ce7-0084-5ad
[jt]  junction integrgation time  0.75
[ct]  chordal tolerance           0.0100 mm
[sl]  soft limit enable           0 [0=disable,1=enable]
[lim] limit switch enable         1 [0=disable,1=enable]
[saf] safety interlock enable     1 [0=disable,1=enable]
[m48e] overrides enabled          1 [0=disable,1=enable]
[mfoe] manual feed override enab  0 [0=disable,1=enable]
[mfo]  manual feedrate override   1.000 [0.05 < mfo < 2.00]
[mtoe] manual traverse over enab  0 [0=disable,1=enable]
[mto]  manual traverse override   1.000 [0.05 < mto < 1.00]
[mt]  motor idle timeout          2.00 seconds
[spep] spindle enable polarity    1 [0=active_low,1=active_high]
[spdp] spindle direction polarity 1 [0=CW_low,1=CW_high]
[spph] spindle pause on hold      1 [0=no,1=pause_on_hold]
[spdw] spindle dwell time         1.0 seconds
[ssoe] spindle speed override ena 0 [0=disable,1=enable]
[sso] spindle speed override      1.000 [0.050 < sso < 2.000]
[cofp] coolant flood polarity     1 [0=low is ON,1=high is ON]
[comp] coolant mist polarity      1 [0=low is ON,1=high is ON]
[coph] coolant pause on hold      1 [0=no,1=pause_on_hold]
[tv]  text verbosity              1 [0=silent,1=verbose]
[ej]  enable json mode            2 [0=text,1=JSON,2=auto]
[jv]  json verbosity              2 [0=silent,1=footer,2=messages,3=configs,4=linenum,5=verbose]
[qv]  queue report verbosity      0 [0=off,1=single,2=triple]
[sv]  status report verbosity     1 [0=off,1=filtered,2=verbose]
[si]  status interval           250 ms
[gpl] default gcode plane         0 [0=G17,1=G18,2=G19]
[gun] default gcode units mode    1 [0=G20,1=G21]
[gco] default gcode coord system  1 [1-6 (G54-G59)]
[gpa] default gcode path control  2 [0=G61,1=G61.1,2=G64]
[gdi] default gcode distance mode 0 [0=G90,1=G91]
g2core[mm] ok> 
----------------------------

[image]

Sending sample program, only SPJS Buffer fills up, board not responding.

[image]

hurr1k4ne, 2017-01-06:
seems to work now....
edited settings.

aldenhart, 2017-01-07:
We have some people having performance problems with Windows10 configurations. Can you provide some more information about what changes you mad to the settings to fix your problem?

hurr1k4ne, 2017-01-08:
Hey Alden,

i am very sorry, but i'm not able to say anything about it. I disassembled the Due board an programmed it with my laptop with Ubuntu 16.10, own config, make, bossac over programming port. then tried it in Win10. doesn't work. 
In the evening i reassembled it in my self made mill and after some problems with configs it worked.
But it was just something with axes and Input config.

Today i milled a quite complex Fusion 360 Gcode with adaptives and 3d horizontals with chilipeppr. (works great)
Only thing i struggle is that when trying G38.2 Z-Probing it directly goes into "shutdown" alarm state.
But i'm almost sure it is a config problem... (Using a electrical "pull-down" PCB probe thingi ;-) )

If i will find anything corresponding the problem i will get in contact with you,
(I am experienced with debugging software and hardware)

hurr1k4ne, 2017-01-12:
closed due to fix in 100.19

closed 2017-01-12 by hurr1k4ne